stereolabs/zed-ros-wrapper #694: Configuring zed 2 with the navigation stack[Question]

MahmoudF98, 2021-03-25:
Hello,

Is there is a tutorial to configure the zed 2 camera with the ROS navigation stack?

Thanks in advance!

Myzhar, 2021-03-25:
Hi @MahmoudF98 
no, there is not such a tutorial available yet.
Please do not hesitate to ask for any info you need.

ahmetsaglam, 2022-03-28:
Hi @Myzhar 
I have questions about ROS navigation stack setup. I have ZED2i camera and a lidar. Lidar provides /scan data. ZED 2i provides odometry. 

Question 1: If I want to run gmapping to map (2d) the environment, I need to turn off map to odom transformation in ZED wrapper as gmapping needs to provide this transformation. However, doing so causes too much odom drift and eventually an incorrect map of the environment. How can I benefit map to odom tf of ZED to map (just 2d) the environment with gmapping or something else?
Question 2: If I already have a map of the environment, I can run amcl to localize the robot in the map. Similar to above, amcl publishes map to odom tf, so I need to turn off ZED's map to odom tf. Is there any to fuse both?
Question 3: Is it possible to use ZED 2i with actual, real-time SLAM? If so, what would be an efficient and effective SLAM algorithm/method use with ZED 2i?

Thanks in advance,
Ahmet

Myzhar, 2022-03-28:
hI @ahmetsaglam 

1. You must configure gmapping (if possible) to use the ZED Wrapper odometry, by disabling the publishing of the `map` -> `odom` transform in all the other nodes
2. Yes, you must use a Kalman filter (see EKF or UKF) in the navigation stack.
3. You can use the SLAM module of the ZED SDK (pose topic) or you can use RTABmap which is the state of the art.

ahmetsaglam, 2022-03-28:
Thank you for quick, concise, and through reply.

Let's assume I will use SLAM module of the ZED SDK for robot's pose. What would be a good algorithm for obstacle avoidance with ZED 2i? 
Normally, ROS Navigation stack expects laser scan for obstacles. However, converting  depth or point cloud to scan data is not a good solution as we do not benefit vision. Do you have any suggestions for obstacle avoidance?

ahmetsaglam, 2022-03-28:
You're right, we can also use point cloud as sensor input. I missed that because usually we do not want to use it with limited resources. 
I will use try this setup and see how it will perform on Jetson Nano and Jetson Xaiver.

I have another question, though:
> external Kalman filter to fuse ZED odometry and other odometry sources (e.g. wheels).
Does this mean we can still use ZED pose topic for map to odom tf but we can fuse ZED odometry with other odometry sources for a better odometry estimate (odom to robot base tf).  Is my understanding correct?

Myzhar, 2022-03-28:
> Does this mean we can still use ZED pose topic for map to odom tf but we can fuse ZED odometry with other odometry sources for a better odometry estimate (odom to robot base tf). Is my understanding correct?

Yes, that's correct

ahmetsaglam, 2022-03-28:
Great, thanks!

mingshi1214, 2022-03-30:
Hi @Myzhar and @ahmetsaglam , 

> Does this mean we can still use ZED pose topic for map to odom tf but we can fuse ZED odometry with other odometry sources for a better odometry estimate (odom to robot base tf). Is my understanding correct?

I found that when I use the ZED pose tf for map to odom, this prevents the amcl to localize the ZED2. Before I modified the zed_camera.launch file to set the param "pos_tracking/publish_map_tf" to be false, the amcl node would try to localize the camera and publish a map to odom tf but the ZED camera would also publish a  tf from map to odom which would cause glitching between the ZED and AMCL positions of the odom frame when I view it in RVIZ. This was fixed after the publish_map_tf param was set to false. I found this in [https://github.com/stereolabs/zed-ros-wrapper/issues/281](url). 

However, I am generally having issues with amcl and ZED. When the amcl sends the predicted tf between map and odom, the position of the ZED moves in RVIS but the local map is not moving position or orientation. Hopefully this is due to the set_pose bug that is being fixed now. 

If this is not due to this bug, would you have a suggestion on how to synchronize the amcl node with the ZED?

Myzhar, 2022-03-31:
If you are using AMCL then it's correct to set `pos_tracking/publish_map_tf` to false.

If the ZED camera correctly moves and the local map doesn't, then there is something to be fixed in the TF chain.

You can use the `roswtf` command to search for problems and examinate the TF tree with `rqt`.

closed 2022-05-07 by github-actions[bot]