stereolabs/zed-ros-wrapper #696: [Question] topic "odom" based on visual info or imu info?

hyylx94, 2021-03-30:
Dear STEREOLABS team, 

I'm using ZED mini and SDK 3.4.2. I have a question about the "odom" topic. 
If I give an SVO file (imu info is recorded) as input, then I launch the ZED node. Is The "odom" topic I get based on the visual information or visual+imu information?

Thanks in advance!

Kind regards,
Xin

Myzhar, 2021-03-30:
Hi @hyylx94 
the `odom` topic is always based on "Visual-Inertial" information, if you are using a ZED2 or a ZED-Mini.
The ZED (first model) only provides visual odometry.

hyylx94, 2021-04-22:
Hi @Myzhar 
Thanks for your reply.
I also would like to ask how did you integrate the visual odometry and imu info? 
    1. How much the odom topic relay on the visual information and how much on the imu infomation?
    2. Will the odom topic switch to inertial only if the environment is dark?

I also did an experiment of blocking the cameras while moving ZEDm (I expect that the odom then will relay only on the inertial info). What I see is when I block the camera, both odom and imu topics are not update (seems like 'forzen'). Do you know the reason for behavior?

Thanks in advance!

closed 2022-04-19 by github-actions[bot]