stereolabs/zed-ros-wrapper #698: How to display full image in zed ros in VGA resolution?

Tranks27, 2021-04-05:
Hi,
Currently I'm trying to subscribe to the left image topic and view it using opencv, as shown below. I can view the whole image with the default resolution of 1280x720 but the problem is when I changed the zed resolution to VGA via the **params/common.yaml** , the image window only shows a small portion of the actual image (see [here](https://imgur.com/a/PbJigHB)). Please help.
 
`void imageCallback(const sensor_msgs::Image::ConstPtr& msg){
    
    cv_bridge::CvImagePtr cv_ptr;
    try{
        cv_ptr = cv_bridge::toCvCopy(*msg, sensor_msgs::image_encodings::BGR8);
    }catch (cv_bridge::Exception& e){
        ROS_ERROR("error %s", e.what());
        return;
    }
    cv::namedWindow("image window", CV_WINDOW_NORMAL);
    cv::resizeWindow("image window", 600,600);
    cv::imshow("image window", cv_ptr->image);
    
    cv::waitKey(3);
    
    
}

int main(int argc, char** argv){
    ros::init(argc, argv, "dumb_image_processor_node");
    ros::NodeHandle nh;

    ros::Subscriber sub = nh.subscribe("/zed2/zed_node/right/image_rect_color", 1, imageCallback);

    ros::spin();    

    return 0;

}`

Myzhar, 2021-05-17:
Hi @Tranks27 
there is an issue with your "viewer".
Please check the `image_encodings` of the incoming topics, you will notice that it's not `BGR8`, but `BGRA8`.

Tranks27, 2021-05-18:
> Hi @Tranks27
> there is an issue with your "viewer".
> Please check the `image_encodings` of the incoming topics, you will notice that it's not `BGR8`, but `BGRA8`.

Hi @Myzhar 
I have changed the encoding to `BGRA8` but the error still persists.

Tranks27, 2021-05-18:
It appears to me that this error happens only when I subscribe to `rectified image` topics. Subscribing to `raw image` topic works fine. Is there any reasons to that?

Myzhar, 2021-05-18:
@Tranks27 can you share your package as a compressed file? 
I could test it for you to see if I find the bug in the wrapper or in the package itself

Tranks27, 2021-05-18:
My package is just a very simple image viewer. [Here](https://drive.google.com/file/d/1vi0_mG1QrEl0b8N6ssLBf2sumQEktKp-/view?usp=sharing) it is.

Myzhar, 2021-05-18:
OK, the problem is in your code. You used the `WINDOWS_NORMAL` flag for `cv::namedWindow`, with this flag you must manually resize the window to fit the image.
Try using automatic resizing with `cv::WINDOW_AUTOSIZE`:

```
void imageCallback(const sensor_msgs::Image::ConstPtr& msg){

    ROS_INFO("Left image received - Size: %dx%d", msg->width, msg->height);
    
    cv_bridge::CvImagePtr cv_ptr;
    try{
        cv_ptr = cv_bridge::toCvCopy(*msg, sensor_msgs::image_encodings::BGRA8);
    }catch (cv_bridge::Exception& e){
        ROS_ERROR("error %s", e.what());
        return;
    }
    cv::namedWindow("image window", cv::WINDOW_AUTOSIZE);
    // cv::resizeWindow("image window", 600,600);
    cv::imshow("image window", cv_ptr->image);
    cv::waitKey(1);   
}

Tranks27, 2021-05-18:
> OK, the problem is in your code. You used the `WINDOWS_NORMAL` flag for `cv::namedWindow`, with this flag you must manually resize the window to fit the image.
> Try using automatic resizing with [...]

The error still persists. Besides, are you able to replicate this error on your pc?

Myzhar, 2021-05-18:
No, on my PC it works like expected. If I change the resolution in `common.yaml`, the "image window" windows changes its size according to it.

Tranks27, 2021-05-18:
I tried it on another laptop and it was working. So I figured something might have been wrong with my zed sdk. So I deleted my old sdk and installed the latest v3.5. It's running ok now. Thanks for all the support on the way.

closed 2021-05-18 by Tranks27