stereolabs/zed-ros-wrapper #699: Configure Sensor rate via configuration

ladrians, 2021-04-06:
Hello, the requst is to configure the desired rate for the sensors via configuration files.
I need to use the IMU (publishing at 400hz) and Magnetometer (50hz) and is too much for our requirements.

It would be useful to define the desired rate by sensor at startup, similar to the PointCloud or camera configuration.
So far, the option has beep to use the [drop](http://wiki.ros.org/topic_tools/drop) utility for example something like

`<node pkg="topic_tools" type="drop" name="zed_imu_relay" args= "/zed/zed_node/imu/data_raw 19 20 /imu/data_raw" />
`
thanks,

Myzhar, 2021-04-06:
Hi @ladrians 
indeed this can be a useful feature.
We can add a variable to throttle the frequency of the publishing of the sensors data as it is available for images and depth streams.
Stay tuned for updates

ladrians, 2022-04-13:
Any news on this request?

Myzhar, 2022-04-13:
The feature has been introduced with 57cc33f2fa4bff709e29c5ff20571118ac3f268f

closed 2022-04-19 by Myzhar