stereolabs/zed-ros-wrapper #700: [Question] usr/bin/ld: warning: libopencv_calib3d.so.3.2, needed by /opt/ros/melodic/lib/libimage_geometry.so, may conflict with libopencv_calib3d.so.3.4

xiaohuozi-code, 2021-04-06:
how can i solve it~~~
thanks~~

Myzhar, 2021-04-06:
Hi @xiaohuozi-code 
Can you provide more information about this issue?
It seems to not being related to a ZED Wrapper problem, but to a misconfiguration of your PC.

xiaohuozi-code, 2021-04-07:
> Hi @xiaohuozi-code
> Can you provide more information about this issue?
> It seems to not being related to a ZED Wrapper problem, but to a misconfiguration of your PC.

/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
[ 98%] Linking CXX executable /home/qupengtao/catkin_ws/devel/lib/pointcloud_mapping/pointcloud_mapping
[100%] Linking CXX executable /home/qupengtao/catkin_ws/devel/lib/pointcloud_mapping/mappingWithPointCloud
/usr/bin/ld: warning: libopencv_flann.so.3.2, needed by /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.3.2.0, may conflict with libopencv_flann.so.3.4
/usr/bin/ld: warning: libopencv_imgcodecs.so.3.2, needed by /opt/ros/melodic/lib/libcv_bridge.so, may conflict with libopencv_imgcodecs.so.3.4
/usr/bin/ld: warning: libopencv_imgproc.so.3.4, needed by /usr/local/lib/libopencv_imgcodecs.so.3.4.13, may conflict with libopencv_imgproc.so.3.2
[100%] Built target pointcloud_mapping
/usr/bin/ld: warning: libopencv_flann.so.3.2, needed by /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.3.2.0, may conflict with libopencv_flann.so.3.4
/usr/bin/ld: warning: libopencv_imgcodecs.so.3.2, needed by /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.3.2.0, may conflict with libopencv_imgcodecs.so.3.4
[100%] Built target mappingWithPointCloud
qupengtao@qupengtao-OMEN-by-HP-Laptop-15-dc1xxx:~/catkin_ws$ 

Maybe I put the problems in wrong place,sorry!
But could you help me?

Myzhar, 2021-05-17:
Can you please update the status of this issue and close it if it's solved?

closed 2021-05-17 by xiaohuozi-code