stereolabs/zed-ros-wrapper #701: [BUG]

2137254d, 2021-04-07:
**Describe the bug**
The visual odometry drifts drastically when an object or person walks across the field of view. We tried to use the visual odometry from the zed 2 camera as we were already using it for pointclouds, but we kept facing the issue that the odometry would drift and then start to fluctuate wildly which is a major issue for navigation.

**Device information:**
 - OS: Ubuntu
 - OS version: 18.04
 - CPU: Jetson Xavier
 - ZED SDK Version: v3.4.2
 - ROS Wrapper version: v3.4.x

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'rviz'
2. Subscribe to 'zed/odom'
3. walk in front of camera a few times at various close distances '....'
4. See error

Myzhar, 2021-04-07:
Hi @2137254d 
thank you for reporting this problem. We will try to replicate and fix it.

However, the odometry topic can drift because it is not fixed by closed-loop checking (odometry cannot jump, see [REP105](https://www.ros.org/reps/rep-0105.html#odom))
I suggest you write a node that monitors the covariance matrix of the odometry and restart it when it increases too much.

Myzhar, 2021-05-07:
@2137254d have you tried also fusing the visual-inertial odometry provided by the ZED2 with other external sources of odometry (e.g. wheel odometry) using a Kalman Filter?
Even if the visual odometry is fused with inertial information by the ZED SDK, drifts are an ineliminable factor and you should take care of them by resetting odometry when the drift is too big (covariance is a good indicator) and by fusing it with other sources.
Furthermore, the ZED SDK provides a more stable estimation of the camera position with the zed/pose topic that is not odometry because the pose can jump when a fix is applied when the algorithm detects a "known position from the past".

FishInWave, 2021-06-24:
> @2137254d have you tried also fusing the visual-inertial odometry provided by the ZED2 with other external sources of odometry (e.g. wheel odometry) using a Kalman Filter?
> Even if the visual [...]

Hi, in[ zed's official doc](https://www.stereolabs.com/docs/ros/positional-tracking/), it says that 

> odometry: The position calculated as the sum of the movements relative to the previous position. Only the pure visual odometry is used
So if I enable IMU Fusion, and use `getPosition(zed_pose, REFERENCE_FRAME::CAMERA)` to get relative pose, is this transform corrected by IMU?

Myzhar, 2021-06-24:
Yes, the "pose" topic is not odometry indeed. It has jumps because of the many corrections made by the SLAM algorithm when a "known position" is identified.

closed 2022-04-19 by github-actions[bot]