stereolabs/zed-ros-wrapper #703: [Question]

oussspo, 2021-04-08:
i finishe my workspace catkin_ws , but when i exc it i get this msg
[image]

Myzhar, 2021-04-08:
Can you post the result of
`ls /home/oussema/catkin_ws`
and
`ls /home/oussema/catkin_ws/src`
It seems that your catkin workspace is not correctly configured

oussspo, 2021-04-08:
i think so because i got an other error that the package could not be found 
[image]

oussspo, 2021-04-08:
to find the source file in the workspace we need just to exc roscd but i got the wrong resultat

Myzhar, 2021-04-09:
@oussspo the command `roscd` works only if your catkin workspace is correctly configured and if you correctly set up the environmental variables.
Have you sourced `catkin_ws/devel/setup.bash` at least one time after compilation?

oussspo, 2021-04-09:
yes i have already done it

Myzhar, 2021-04-09:
Then there is something wrong with your CMake cache.
Delete the folders `build` and `devel` in your catkin workspace and delete the file `CMakeLists.txt` in `catkin_ws/src`.
Next go back to `catkin_ws` and start a new compilation process with `catkin_make`

oussspo, 2021-04-09:
I solved the problem, now i had a new
[image]

Myzhar, 2021-04-09:
I'm sorry but this is not strictly related to the ZED ROS Wrapper and I cannot help you with it.
Try to post the error on the [ROS Answers website](https://answers.ros.org/), you will surely find someone that can help you in a better way.

closed 2021-04-09 by Myzhar