stereolabs/zed-ros-wrapper #704: [Question] Publish only image topic on ROS

NamDinhRobotics, 2021-04-09:
Hi,
I want to use multiple Zed 2, when I run zed_wrap node for each zed 2,
It publishes every topic of the cameras: images, PCL, ... too many topics are published, and waste PC resources too much, 

So, I just want to publish only images on ROS,
Please help me solve the problem!

Myzhar, 2021-04-09:
Hi @NamDinhRobotics 
the ZED wrapper advertises the topics, but it does not publish what is not subscribed.

However, you can disable depth elaborations simply by setting these parameters:
```
depth/quality: 0
depth/depth_stabilization: 0
pos_tracking/publish_tf: false
pos_tracking/publish_map_tf: false
```

NamDinhRobotics, 2021-04-09:
Hi @Myzhar 
Thank you for your help,
I mean having any method to just publish the only images without any computation for state estimation or else,...
just like other simple cameras.

Myzhar, 2021-04-09:
You must rewrite your "simple" node that publishes only the data that you want.
This is a "wrapper" for the ZED SDK, so it exposes all the features available in the ZED SDK, but it enables only what you set to be enabled.

NamDinhRobotics, 2021-04-09:
Ok, I will try,
Thank you for your supports,

closed 2022-04-19 by github-actions[bot]