stereolabs/zed-ros-wrapper #705: Point cloud accuracy decreases with camera_flip = True [BUG]

orrblue, 2021-04-14:
I see an issue where the mapping fused pointcloud accuracy diminishes and the sky becomes part of the pointcloud. This seems to happen when our camera is upside down and camera_flip = True.

Our camera is always upside down during mapping, and initially, we were using camera_flip = False and visualized the mapping/fused_cloud pointcloud. It looks good, with just small regions occasionally showing the sky. I expect we can fix that by playing with the depth confidence. 

However, when camera_flip = True the pointcloud quality is really bad, and a lot of the sky appears in the pointcloud. See the image below. I think that's a bug, but happy to hear ideas on how to solve this.

**Device information:**
 - OS: L4T / Jetpack
 - CPU: Jetson Xavier NX
 - ZED SDK Version: 3.4.2

These images contain the mapping/fused_cloud as well as some meshing (provided by ROS package voxblox) but please disregard the mesh. As you can see most of the visualization here is purely the pointcloud. The sky I refer to are all the blue points.

[image]

[image]

This is how much sky we get with camera_flip = False. It's only in the top left of the building. Camera is mounted upside-down as before. Much less sky shows up in the pointcloud. Note that the points here were visualized with a larger size so they may appear as a mesh, but its still a pointcloud:

[image]

Here's the common.yaml:

# Dynamic parameters cannot have a namespace
brightness:                 4                                   # Dynamic
contrast:                   4                                   # Dynamic
hue:                        0                                   # Dynamic
saturation:                 4                                   # Dynamic
sharpness:                  4                                   # Dynamic
gamma:                      8                                   # Dynamic - Requires SDK >=v3.1
auto_exposure_gain:         true                                # Dynamic
gain:                       100                                 # Dynamic - works only if `auto_exposure_gain` is false
exposure:                   100                                 # Dynamic - works only if `auto_exposure_gain` is false
auto_whitebalance:          true                                # Dynamic
whitebalance_temperature:   42                                  # Dynamic - works only if `auto_whitebalance` is false
depth_confidence:           50                                  # Dynamic
depth_texture_conf:         100                                 # Dynamic
pub_frame_rate:             5.0                                # Dynamic - frequency of publishing of video and depth data
point_cloud_freq:           15.0                                # Dynamic - frequency of the pointcloud publishing (equal or less to `grab_frame_rate` value)

general:
    camera_name:                zed                             # A name for the camera (can be different from camera model and node name and can be overwritten by the launch file)
    zed_id:                     0
    serial_number:              0
    resolution:                 2                               # '0': HD2K, '1': HD1080, '2': HD720, '3': VGA
    grab_frame_rate:            30                              # Frequency of frame grabbing for internal SDK operations
    gpu_id:                     -1
    base_frame:                 'base_link'                     # must be equal to the frame_id used in the URDF file
    verbose:                    false                           # Enable info message by the ZED SDK
    svo_compression:            2                               # `0`: LOSSLESS, `1`: AVCHD, `2`: HEVC
    self_calib:                 true                            # enable/disable self calibration at starting
    camera_flip:                false

video:
    img_downsample_factor:      1.0                             # Resample factor for images [0.01,1.0] The SDK works with native image sizes, but publishes rescaled image.
    extrinsic_in_camera_frame:  true                            # if `false` extrinsic parameter in `camera_info` will use ROS native frame (X FORWARD, Z UP) instead of the camera frame (Z FORWARD, Y DOWN) [`true` use old behavior as for version < v3.1]

depth:
    quality:                    3                               # '0': NONE, '1': PERFORMANCE, '2': QUALITY, '3': ULTRA
    sensing_mode:               0                               # '0': STANDARD, '1': FILL (not use FILL for robotic applications)
    depth_stabilization:        1                               # `0`: disabled, `1`: enabled
    openni_depth_mode:          false                           # 'false': 32bit float meters, 'true': 16bit uchar millimeters
    depth_downsample_factor:    1.0                             # Resample factor for depth data matrices [0.01,1.0] The SDK works with native data sizes, but publishes rescaled matrices (depth map, point cloud, ...)

pos_tracking:
    publish_tf:                 true                            # publish `odom -> base_link` TF
    publish_map_tf:             true                            # publish `map -> odom` TF
    map_frame:                  'map'
    odometry_frame:             'odom'
    area_memory_db_path:        ''
    area_memory:                true                            # Enable to detect loop closure
    floor_alignment:            false                           # Enable to automatically calculate camera/floor offset
    initial_base_pose:          [0.0,0.0,0.0, 0.0,0.0,0.0]      # Initial position of the `base_frame` -> [X, Y, Z, R, P, Y]
    init_odom_with_first_valid_pose: true                       # Enable to initialize the odometry with the first valid pose
    path_pub_rate:              2.0                             # Camera trajectory publishing frequency
    path_max_count:             -1                              # use '-1' for unlimited path size
    two_d_mode:                 false                           # Force navigation on a plane. If true the Z value will be fixed to "fixed_z_value", roll and pitch to zero
    fixed_z_value:              0.00                            # Value to be used for Z coordinate if `two_d_mode` is true

mapping:
    mapping_enabled:            true                           # True to enable mapping and fused point cloud pubblication
    resolution:                 0.04                            # maps resolution in meters [0.01f, 0.2f]
    max_mapping_range:          7                              # maximum depth range while mapping in meters (-1 for automatic calculation) [2.0, 20.0]
    fused_pointcloud_freq:      2                             # frequency of the publishing of the fused colored point cloud

Myzhar, 2021-04-15:
Hi @orrblue 
this issue with the flip mode is indeed weird and we need to investigate it further.

Can I ask you to record an SVO for both flip mode enabled and disabled and provide them to us sending a download link (dropbox, google drive, ecc) to support@stereolabs.com?

orrblue, 2021-04-16:
Thanks for the quick reply. I'll try doing that the next time I go out to do mapping, which could be next week.

closed 2022-04-19 by github-actions[bot]