stereolabs/zed-ros-wrapper #706: [BUG]

zhangliyun9120, 2021-04-15:
**Problem:**
I used:
$ roslaunch zed_display_rviz display_zedm.launch
$ rosservice call /zedm/zed_node/start_svo_recording test.svo
got error:
[ INFO] [1618486893.935390804]: SVO recording STARTED: test.svo (H265 (HEVC))
[ERROR] [1618486893.947473225]: Error saving frame to SVO

**My environment:**
Ubuntu 16.04
cuda_10.0.130_410.48_linux.run 
ZED_SDK_Ubuntu16_cuda10.0_v3.4.2.run

**shotscreen**
[image]

zhangliyun9120, 2021-04-16:
@Myzhar Sorry, 
Can you push this issue as soon as possible, since we have an important project debugging was blocked as this issue. And my ros version is kinetic, which maybe is also a reason?
 If there are some process pls let me know, Thank you so much!

Myzhar, 2021-04-16:
Have you tried changing the encoder to h264? Maybe that your GPU card does not support HW H265 encoding

zhangliyun9120, 2021-04-16:
> Have you tried changing the encoder to h264? Maybe that your GPU card does not support HW H265 encoding

I never do this, can I use a command parameter to do it when I using rosserice command for svo format video saving?
e.g. :  $  rosservice call /zedm/zed_node/start_svo_recording test.svo --xxx
can you check this command for me? Thank you so much.

Myzhar, 2021-04-16:
You can set it in the `common.yaml` parameter file:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L31
change it to "1" that is H264

zhangliyun9120, 2021-04-17:
> You can set it in the `common.yaml` parameter file:
> https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L31
> change it to "1" that is H264

Hi Myzhar,
I tried this as your solution, but it sill got the error as follows:

[image]

It looks like already as H264 but not work. Do you have another solution? 
Thank you.

Myzhar, 2021-04-17:
I'm sorry, but I can't see the error in the screenshoot.
What's it?

zhangliyun9120, 2021-04-17:
> I'm sorry, but I can't see the error in the screenshoot.
> What's it?

what do you mean?
As shown in screenshot, it is still occurred:

"SVO recording STARTED: test.svo (H264 (AVCHD))
[ERROR] Error saving frame to SVO"

closed 2022-04-19 by github-actions[bot]