stereolabs/zed-ros-wrapper #712: > 安装zed，运行ZED_Dignosizic,诊断相机连接时：processor

shangshanruoshui123, 2021-04-28:
> 安装zed，运行ZED_Dignosizic,诊断相机连接时：processor
> 报错：unable to identify your processor,make sure that it fits the requirements；

请问你这个问题解决了吗？我也遇到了无法识别processor的问题

_Originally posted by @zhoushuailin in https://github.com/stereolabs/zed-ros-wrapper/issues/686#issuecomment-803230125_

zhoushuailin, 2021-04-28:
我也没有解决，目前使用基本的数据采集没有问题，但是在运行自带的深度图像和svo等算法就会显示无法识别cuda.

发自我的iPhone

  

  
 
Originally posted by @zhoushuailin in #686 (comment)

shangshanruoshui123, 2021-04-28:
我刚刚还发现一个问题，运行nvcc-V命令，都反应未找到该命令,每次都需要重新更新配置环境，很奇怪，不晓得是不是安装英伟达系统时发生什么错误。

zhoushuailin, 2021-04-28:
是的，我似乎也遇到过这样的问题，目前来看比较正常。
但是我的不是英伟达的显卡，是amd的显卡

发自我的iPhone

 
 
 
 
 ------------------&amp;nbsp;原始邮件&amp;nbsp;------------------ 
 主题:&amp;nbsp;Re: [stereolabs/zed-ros-wrapper] &amp;gt; 安装zed，运行ZED_Dignosizic,诊断相机连接时：processor (#712)

shangshanruoshui123, 2021-04-28:
我想我知道你的问题，zed sdk 需要安装cuda，而cuda需要英伟达显卡才可以。希望能帮到你。

 
 
 
 
 
 
 
 
   
   
   
   
  ------------------&amp;amp;nbsp;原始邮件&amp;amp;nbsp;------------------  
  主题:&amp;amp;nbsp;Re: [stereolabs/zed-ros-wrapper] &amp;amp;gt; 安装zed，运行ZED_Dignosizic,诊断相机连接时：processor (#712)

zhoushuailin, 2021-04-28:
cuda不一定要英伟达的，这个显卡装好驱动,cuda以后，能正常运行gpu版本的yolo3

发自我的iPhone

 
 
 
 
 
 
 
   
   
   
   
   
   
   
   
     
     
     
     
   ------------------&amp;amp;amp;nbsp;原始邮件&amp;amp;amp;nbsp;------------------   
   主题:&amp;amp;amp;nbsp;Re: [stereolabs/zed-ros-wrapper] &amp;amp;amp;gt; 安装zed，运行ZED_Dignosizic,诊断相机连接时：processor (#712)

shangshanruoshui123, 2021-04-28:
NVIDIA公司发布了CUDA(Compute Unified Device Architecture)，是一种新的操作GPU计算的硬件和软件架构，是建立在NVIDIA的GPUs上的一个通用并行计算平台和编程模型。我的理解是，cuda只能在nvidia显卡运行。

 
 
 
 
 
 
 
 
 
 
   
   
   
   
   
   
   
     
     
     
     
     
     
     
     
       
       
       
       
    ------------------&amp;amp;amp;amp;nbsp;原始邮件&amp;amp;amp;amp;nbsp;------------------    
    主题:&amp;amp;amp;amp;nbsp;Re: [stereolabs/zed-ros-wrapper] &amp;amp;amp;amp;gt; 安装zed，运行ZED_Dignosizic,诊断相机连接时：processor (#712)

Myzhar, 2021-04-28:
@shangshanruoshui123 @zhoushuailin can you please post using the English language? 
It will be easier for all the users to read you questions and reply to them

shangshanruoshui123, 2021-04-28:
Hi, I installed zed sdk(4.5 or 4.4) on my Nvidia nano(jetpack4.4,cuda10.2), but I can’t run ZED_Depth_Viewer, and Diagnostic shows Unable to indentify your processors.
       I am looking forward to your help .

Myzhar, 2021-04-28:
@shangshanruoshui123 I'm sorry, but this is a repository related to robotics with ROS.
Can you please send an email to support@stereolabs.com describing your system's configuration and attaching the report generated by the ZED Diagnostic tool?
You will get a fast and detailed support

closed 2021-04-28 by Myzhar

459737087, 2022-04-21:
Hello! How to deal with this problem?
I also met this problem,
the cuda version 11.5
cudnn 8.3.3
ubuntu 20.04

Myzhar, 2022-04-21:
Hi @459737087 
please do not reply to closed issues. As I wrote one comment above:

I'm sorry, but this is a repository related to robotics with ROS.
Can you please send an email to [support@stereolabs.com](mailto:support@stereolabs.com) describing your system's configuration and attaching the report generated by the ZED Diagnostic tool?
You will get a fast and detailed support

459737087, 2022-04-21:
ok Thank you @Myzhar

lymcs, 2022-12-16:
hello,excuse me. I met a problem:Unable to indentify your processor，How did you solve it??

lymcs, 2022-12-16:
> I have the same problem: Unable to indentify your processor，How did you solve it
> 
> > 安装zed，运行ZED_Dignosizic,诊断相机连接时：processor
> > 报错：unable to identify your processor,make sure that it fits the requirements；
> 
> 请问你这个问题解决了吗？我也遇到了无法识别processor的问题
> 
> _Originally posted by @zhoushuailin in [#686 (comment)](https://github.com/stereolabs/zed-ros-wrapper/issues/686#issuecomment-803230125)_

兄弟，解决没？？无法识别处理器？？

Myzhar, 2022-12-16:
> Can you please send an email to [support@stereolabs.com](mailto:support@stereolabs.com) describing your system's configuration and attaching the report generated by the ZED Diagnostic tool?
> You will get a fast and detailed support