stereolabs/zed-ros-wrapper #713: Problem extrapolating /zed2/zed_node/obj_det/objects from rosbag

Baldins, 2021-04-29:
Hi, 

I used rosbag to record /zed2/zed_node/obj_det/objects but when I try to extrapolate the recording and export into the csv file I get all data saved in a single column and I am not sure how to get them and separate label, ids bounding boxes etc.

Any chance do you have any tips for solving this problem? 

Thank you!

Myzhar, 2021-04-29:
Hi @Baldins 
How do you export the topic to CSV?

Baldins, 2021-04-29:
This is my code: 

`import bagpy
from bagpy import bagreader
import pandas as pd

bag = bagreader("2021-03-30-18-24-59.bag")
topic = '/zed2/zed_node/obj_det/objects'

objects = bag.message_by_topic(topic)
df_objects = pd.read_csv(objects)`

Am I doing something wrong here?

Myzhar, 2021-04-29:
The `/zed2/zed_node/obj_det/objects` topic is complex, it contains an array of object sub-topics with many other sub-topics.
You must be sure that the topic is fully parsed when exported to CSV.
The best thing is to write your own node that parses each field and subfield of the `ObjectStamped` topic.
Here you can find more information:
- https://github.com/stereolabs/zed-ros-wrapper/tree/master/zed_interfaces/msg
- https://www.stereolabs.com/docs/ros/object-detection/#detected-objects-message

Baldins, 2021-04-29:
Thank you! 

I am having some problem with the parsing part though. The examples I saw are using "import rosbag" but it keeps giving me errors. But with bagpy I am a bit stuck :/ 
Any tips for that too?

Thanks a lot for your help!

Myzhar, 2021-04-29:
I'm sorry, but I never performed that kind of operation. I guess that by asking to "ROS answers" you will get a more precise reply: https://answers.ros.org/

Baldins, 2021-04-30:
I can't find any way to recover the subtopics and subsubtopics from the bag file - you don't really have any tips for that? 

Is there any other way to store the subtopics in a more accessible way for the postprocessing? 

Thank you!

Baldins, 2021-04-30:
Here is the code you can use to visualize the output. 

I am not sure how to get those info and store into a panda dataframe 

> from configparser import ConfigParser
> import rosbag
> bag = rosbag.Bag("2021-03-30-18-24-59.bag")
> 
> bagContents = bag.read_messages()
> parser = ConfigParser()
> 

> for topic, msg, t in bagContents:
>     print(t)
>     print(msg.objects

The csv file is too big and does not allow me to upload it :/ 

Thank you!

closed 2022-04-19 by github-actions[bot]