stereolabs/zed-ros-wrapper #714: Using ZED2 with robot_localization, how to manage base_link?

clint-qurrent, 2021-05-04:
I have a ZED2 that I'd like to leverage the odometry from and compare fusing that with the IMU on the ZED and an IMU+GPS unit that I have. The problem I believe I'm running into is that both `robot_localization` and `zed_wrapper` generate `base_link`. I am using the ` zed_no_tf.launch` so `odom` shouldn't be an issue. Is there a way to force the ZED to use the `base_link` that is generated by `robot_localization`?

Thank you for any insight you can provide.

Myzhar, 2021-05-04:
Hi @clint-qurrent 
the ZED ROS Wrapper internally uses the default TF chain `map`->`odom`->`base_link` described in the [REP105 document](https://www.ros.org/reps/rep-0105.html).
When you disable the odometry broadcasting (like `zed_no_tf.launch` does) the `odom`->`base_link` transform is automatically referred to the transform published by other nodes.
To compare the odometry published by the ZED with the odometry published by other nodes you must be sure that the initial conditions are the same, so you must set the initial ZED position using the `set_pose` service when you receive the first valid odometry pose by an external localization node.

clint-qurrent, 2021-05-04:
@Myzhar 

Ok, so I think the set_pose was the only step I was missing.

Is it possible to automate the set_pose via a launch file or parameter file?

Thank you,
Clint

closed 2022-04-19 by github-actions[bot]