stereolabs/zed-ros-wrapper #715: [Question] where are the /obj_det/object_markers ?

jaronski, 2021-05-05:
i would like to visualize the Markers on the zed2 pointcloud as it was possible in previous version in RVIZ. Am i missing the topic? was the object_marker removed on purpose?

Myzhar, 2021-05-05:
Hi @jaronski 
we removed that topic because it was computationally expensive.
To visualize the result of the object detection with Rviz we instead added a plugin that subscribes to the detected objects topic and renders all the results. 
The plugin is available in the [zed-ros-examples repository](https://github.com/stereolabs/zed-ros-examples/tree/master/rviz-plugin-zed-od).
You can get more information by reading the Stereolabs online documentation:
https://www.stereolabs.com/docs/ros/object-detection/

I hope you will enjoy the improved object detection module, please do not hesitate to provide us with your feedback.

jaronski, 2021-05-05:
Wow this really cool! i run the zed_wrapper on a Xavier AGX and rviz visualization plugin on a different PC... fantastic ! exactly what i needed

jaronski, 2021-05-05:
I noticed this ERROR 
[ERROR] [1620223449.543542856]: PluginlibFactory: The plugin for class 'rviz_imu_plugin/Imu' failed to load.  Error: According to the loaded plugin descriptions the class rviz_imu_plugin/Imu with base class type rviz::Display does not exist. Declared types are  rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped rviz_plugin_tutorials/Imu rviz_plugin_zed_od/ZedOdDisplay
and this Warning:
[ WARN] [1620223450.028346157]: Messages of type 0 arrived closer (0.133160000) than the lower bound you provided (0.500000000) (will print only once)
[ WARN] [1620223495.905411278]: Messages of type 1 arrived closer (0.067041000) than the lower bound you provided (0.500000000) (will print only once)

do you have a solution ?

Myzhar, 2021-05-05:
Sure:
`$ sudo apt install ros-noetic-rviz-imu-plugin`
replace `noetic` with your current ROS version

closed 2022-04-13 by jaronski