stereolabs/zed-ros-wrapper #716: [BUG] ROS TF transforms and P matrix apply transform twice when projecting onto right camera

rmaronson, 2021-05-06:
**Describe the bug**
I am attempting to project a 3D point into a supplied image using ROS TF. I am using the zed wrapper along with the URDF to publish appropriate transforms. 
Say I have a frame `robot_frame` and a published transform between `robot_frame` and `base_link`. Given a point `X` in `robot_frame`, to get its pixel location in the _right_ camera, I can use the following pipeline:

1. Use ROS TF to obtain X in the frame `zed_right_camera_optical_frame` (this is the frame given by both the CameraInfo message and the frame_id of the header in the sensor_msgs/Image provided)
2. Use the K matrix provided by /zed_node/right/camera_info to project that vector into its image frame.

However: by default, ROS's PinholeImageModel uses the full `P` matrix for projection. The CameraInfo message on /zed_node/right/camera_info includes a translation component in `P`, which represents the offset between `zed_left_camera_optical_frame` and `zed_right_camera_optical_frame`. Therefore, if we use the default ROS implementation for the projection, we apply the offset transformation **twice**: once while performing the TF lookup, and again while applying the projection matrix.

Either one of the transformations could be eliminated as follows:
1. Remove the translation component of the P matrix of /zed_node/right/camera_info (may interfere with other stereo applications)
2. Set the frame_id of /zed_node/right/camera_info and /zed_node/right/image headers to the same value as their left counterparts (likely `zed_left_camera_optical_frame`) and rely on the projection matrix to align the frames

**Expected behavior**
Using ROS TF and ROS camera processing information, 3D points should be correctly projected into the right image.

**Additional context**
Hopefully this makes sense -- this took me a lot of digging to find the source of the strange offset, and it's possible I'm misunderstanding something about how this is meant to be used. I haven't been able to find any information about the meaning of the `frame_id` field in either CameraInfo or the sensor_msgs/Image header with respect to appropriate behavior for stereo cameras. But as I understand it, the projection matrix only makes sense when two cameras are referenced from the _same_ frame, and here they are assigned different frames.

Myzhar, 2021-05-06:
Hi @rmaronson 
thank you for submitting this interesting issue. You are right, there is a discrepancy and that's something not easy to solve.
Reasoning quickly on what you noticed I can say that solution 2 is better, but we must think about what this modification can cause.

To fix this weird behavior we should change [this line](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp#L877) from
`mRightCamOptFrameId = mCameraName + "_right_camera_optical_frame";`
to
`mRightCamOptFrameId = mLeftCamOptFrameId";`
in this way the modification that you suggest will be propagated in the whole code.

Can you try it to see if it works like you expect?

rmaronson, 2021-05-06:
Yes, that change does fix it.

Before change (using RViz Camera display)
[image]

After change:
[image]

closed 2022-04-19 by github-actions[bot]