stereolabs/zed-ros-wrapper #717: What is the real shutter speed

RenanGeorgio, 2021-05-10:
Anyone know how to tell me what are the real values ​​of shutter speed of the camera?

At least the range? MIN shutter speed and MAX shutter speed?

Zed only provide de scale from 1-100

Myzhar, 2021-05-10:
@RenanGeorgio I've seen that you cross-posted this question on all the Stereolabs Github repositories.
I suggest you close all the issues on all the Github repositories and send an email to support@stereolabs.com.
Cross-posting on Github, spamming a question, does not follow the good netiquette

RenanGeorgio, 2021-05-12:
> @RenanGeorgio I've seen that you cross-posted this question on all the Stereolabs Github repositories.
> I suggest you close all the issues on all the Github repositories and send an email to [support@stereolabs.com](mailto:support@stereolabs.com).
> Cross-posting on Github, spamming a question, does not follow the good netiquette

I tried.
they didn't give me the answer.
is a simple question...

Myzhar, 2021-05-17:
The exposition time mapping is depending on the framerate.
15fps setExposure(100) -> 19.97ms
30fps setExposure(100) -> 19.97ms
60fps setExposure(100) -> 10.84072ms
100fps setExposure(100) -> 10.106624ms
The exposition is mapped linearly in a percentage of the max value above.
Special case for the setExposure(0) that always corresponds to 0.17072ms

closed 2021-05-17 by Myzhar