stereolabs/zed-ros-wrapper #720: [Question] Dockerfile for zed-ros-wrapper

Leovilhena, 2021-05-14:
Hi, I've been struggling for containerise the zed-ros-wrapper and run it within our stack.

After trying hard I got one built image but got this error:
[ERROR] [1620999145.297092675]: Failed to load nodelet [/zed2/zed_node] of type [zed_nodelets/ZEDWrapperNodelet] even after refreshing the cache: Failed to load library /catkin_ws/devel/lib//libZEDNodelets.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libv4l2.so: cannot open shared object file: No such file or directory)
[ERROR] [1620999145.297360870]: The error before refreshing the cache was: Failed to load library /catkin_ws/devel/lib//libZEDNodelets.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libv4l2.so: cannot open shared object file: No such file or directory)

I've looked for other issues and none of the solutions worked. My container was created from "nvidia/cuda-arm64:11.3.0-runtime-ubuntu20.04" image, with ZED 3.5 SDK, CUDA 11.3 and ROS Noetic installed inside the Dockerfile.

What I need is a arm64 container that would have the same environment settings as a Jetson Xavier NX for running the zed-ros-wrapper and launching zed_wrapper zed2.

Any ideas or resources?

Leovilhena, 2021-09-09:
@Myzhar any insights on this matter?

Myzhar, 2021-09-09:
Hi @Leovilhena 
`Poco exception = libv4l2.so: cannot open shared object file: No such file or directory`
it seems that Video4Linux is not available in your container.
Usually it's installed by the commands
```
sudo apt update
sudo apt install libv4l-0 libv4l-dev v4l-utils 
```

CircleOnCircles, 2021-09-15:
> Hi @Leovilhena
> `Poco exception = libv4l2.so: cannot open shared object file: No such file or directory`
> it seems that Video4Linux is not available in your container.
> Usually it's installed by the commands
> 
> ```
> sudo apt update
> sudo apt install libv4l-0 libv4l-dev v4l-utils 
> ```

This worked.

closed 2022-04-19 by github-actions[bot]