stereolabs/zed-ros-wrapper #721: [Question] Issues with /tf message publishing

bcfdkl-150, 2021-05-17:
I'm using ZED2 and SDK3.5.0. My problem is the /tf message dosen't update automatically, while the other topics like zed2/zednode/odom can update when moving the camera. I can only update the /tf message through restarting rostopic echo /tf  after rostopic echo zed2/zednode/odom. But this will only get one /tf message and get stucked again.  
The issue #408 said that rostpoic echo /tf is the wrong way to debug with tf messages. But my program was sopposed to use the odom messages in /tf. Is there any way to make the /tf message update like the /odom message? Or the /tf message was only supposed to work in this way?  Thx.

Myzhar, 2021-05-17:
The `/tf` topic in ROS does not work like any other topic. 
The correct way to retrieve TF information is using a `transform_listener`, like described in this tutorial:
http://wiki.ros.org/tf/Tutorials/Writing%20a%20tf%20listener%20%28C%2B%2B%29

bcfdkl-150, 2021-05-17:
@Myzhar  Thx a lot for your reply. 
So, if i'm using some program to get the odom information in /tf , and publish it out (for example as /pose). Is the new topic /pose that I get from /tf supposed to not update automaticlly just as the /tf ?  
Maybe I should learn more about the TF information, or just use the zednode/odom message instead.

Myzhar, 2021-05-19:
Yes, they use the same information, but for different scopes

closed 2021-05-28 by Myzhar