stereolabs/zed-ros-wrapper #722: [BUG] Failed to load nodelet [/zed2/zed_node] libcuda.so.1: cannot open shared object file: No such file or directory

kakareko, 2021-05-17:
roslaunch zed_wrapper zed2.launch 
fails to load /zed2/zed_node nodelet with error message:

[ERROR] [1621235098.427662833]: Failed to load nodelet [/zed2/zed_node] of type [zed_nodelets/ZEDWrapperNodelet] even after refreshing the cache: Failed to load library /home/robot/catkin_ws/devel/lib//libZEDNodelets.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libcuda.so.1: cannot open shared object file: No such file or directory)
[ERROR] [1621235098.427689066]: The error before refreshing the cache was: Failed to load library /home/robot/catkin_ws/devel/lib//libZEDNodelets.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libcuda.so.1: cannot open shared object file: No such file or directory)

Could you please advise on how to resolve the issue?

**Device information:**
 - OS: Ubuntu
 - OS version: 18.04
 - CPU: intel i7
 - GPU Nvidia:  GTX1660 Ti Mobile
 - ZED SDK Version: v3.5.0
 - ROS Wrapper version:  latest master
 - ROS Version: Melodic

**To Reproduce**
Steps to reproduce the behavior:
1. Install ZED SDK: ZED_SDK_Ubuntu18_cuda10.2_v3.5.0.run was used for the installation. (Downloaded it from https://download.stereolabs.com/zedsdk/3.5/cu102/ubuntu18)
Note: CUDA Version 10.2.89 existed on the host before the installation of ZED SDK

2. Compile zed-ros-wrapper and zed-ros-examples
cd ~/catkin_ws/src
git clone https://github.com/stereolabs/zed-ros-wrapper.git
git clone https://github.com/stereolabs/zed-ros-examples.git
cd ../
rosdep install --from-paths src --ignore-src -r -y
catkin_make -DCMAKE_BUILD_TYPE=Release
source ./devel/setup.bash

3. try to launch ZED2 ROS wrapper
roslaunch zed_wrapper zed2.launch 

4. See error message:
[ERROR] [1621235098.427662833]: Failed to load nodelet [/zed2/zed_node] of type [zed_nodelets/ZEDWrapperNodelet] even after refreshing the cache: Failed to load library /home/robot/catkin_ws/devel/lib//libZEDNodelets.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libcuda.so.1: cannot open shared object file: No such file or directory)
[ERROR] [1621235098.427689066]: The error before refreshing the cache was: Failed to load library /home/robot/catkin_ws/devel/lib//libZEDNodelets.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libcuda.so.1: cannot open shared object file: No such file or directory)

**Expected behavior**
ZED ROS wrapper works and sends the data into ROS topics.

Myzhar, 2021-05-17:
Hi @kakareko 
the error is self-explaining: CUDA is not correctly initialized on your system and the ZED ROS Wrapper cannot load `libcuda.so.1` while starting.

Can you try to run the following command and paste here the results?:
* `$ nvidia-smi`
* `nvcc -V`

Can you also try to launch `ZED_Explorer` and `ZED_Depth_Viewer` to verify that they can correctly run?

kakareko, 2021-05-17:
Hi @Myzhar 
Thank you for the clarification! I have reinstalled CUDA. The wrapper works fine now.

closed 2021-05-17 by kakareko