stereolabs/zed-ros-wrapper #723: Twist from Odom topic is not updating

dineshrboson, 2021-05-19:
hello 
when I ran zed wrapper and moving my camera on a vehicle at a constant speed and tried to echo the odom and cannot see the twist is changing

Myzhar, 2021-05-19:
Hi @dineshrboson 
the twist information is not yet available. There is an open feature request about it (see https://github.com/stereolabs/zed-ros-wrapper/issues/434), but we have not yet found a good stable solution to implement it.
There are a few challenging problems to be solved that regard the coordinate transformation from camera link to base link.

dineshrboson, 2021-05-19:
yes you can do simple 
speed = distance /time
at every timestamp right?

dineshrboson, 2021-05-19:
ok thanks for the information will close this

closed 2021-05-19 by dineshrboson