stereolabs/zed-ros-wrapper #724: [Question]  Why dose the Zed2 always loss frame when i use "rosbag record" command?

ahuzsg, 2021-05-26:
the rate of camera node is 15,but actually not ,for example: record the /zed2/zed_node/left/image_rect_color for 120s,so there should be 1800 pic, but there just are 1300+ pic, and the time for working longger,loss more

Myzhar, 2021-05-26:
@ahuzsg please provide more info about your system

ahuzsg, 2021-05-26:
help! our car works with the processer: Jetson AGX XAVIER, ubuntu16.04. 
 i found that when i record the VGA pic,it wont loss, but if i record higher rate pix of pic like the 720, the problem appears.

> @ahuzsg please provide more info about your system

ahuzsg, 2021-05-26:
and we use USB3.1 of I/O

Myzhar, 2021-05-26:
The causes of frame loss are two:
1) slow write access to record memory
2) high computational load
rosbag is not well optimized, and it requires a lot of computational power, mainly if you also use compression

ahuzsg, 2021-05-26:
thank you, I guess so, but is there any other way to record pic
from ZED2, write a subscriber?(it may be slower). I'm a novice and this problem has been bothering me for a long time

> The causes of frame loss are two:
> 
> 1. slow write access to record memory
> 2. high computational load
> 
> rosbag is not well optimized, and it requires a lot of computational power, mainly if you also use compression

Myzhar, 2021-05-26:
If you must record only ZED2 data I suggest you use our native recording system to save an SVO file.
The SVO file can then be used as input for the ZED Wrapper to simulate a connected camera.

The only problem with using the SVO method is that you cannot record other topics from external nodes.

For more info:
https://www.stereolabs.com/docs/ros/zed-node/#services
https://www.stereolabs.com/docs/video/recording/

ahuzsg, 2021-05-26:
> If you must record only ZED2 data I suggest you use our native recording system to save an SVO file.
> The SVO file can then be used as input for the ZED Wrapper to simulate a connected camera.
>
> The [...]

thank you ,i'll try it

closed 2021-05-28 by Myzhar