stereolabs/zed-ros-wrapper #725: [Question] Why floor plane publication was removed from wrapper?  How to implement it without forking?

nickzherdev, 2021-05-26:
I saw a page with the code example, but for now I don't see a way how to implement it as a new module. I'd like not to change ros wrapper code. 
https://www.stereolabs.com/docs/spatial-mapping/plane-detection/

Myzhar, 2021-05-26:
Hi @nickzherdev 
the ZED Wrapper never published a topic with information about the floor position.
Floor detection can be enabled to estimate the initial position of the camera with respect to the floor, but no other feature is available.

Can you explain in details what kind of information you are trying to get? Maybe we can add an "action" to be called to retrieve the floor position by command

nickzherdev, 2021-05-26:
Thank you for a quick answer!
Indeed, floor publication was something that my colleagues added after forking a long time ago.

What we are trying to get is a normal to the floor plane or an equation to the floor plane, so that we can dynamically re-calibrate our other cameras. I suppose, the preferred way for such information is a topic with lazy subscription, but an action or a service would be also helpful.

Myzhar, 2021-05-26:
Floor detection is high demanding processing and performing it continuously will reduce the performances of the ROS wrapper. That's why I spoke about an action or a service. 
We will add this feature with one of the next releases of the wrapper. Stay tuned

tcerqueira, 2022-01-10:
I also need a way to get the floor plane, what's the best to achieve this? In the documentation I found the example of "findFloorPlane" in the Camera class but I'm not if this can be used in ros. (Note that I'm getting started with the zed camera)

Myzhar, 2022-01-11:
I think that an action or a service is the right way to provide this feature. I'm sorry, but we had no time to add it yet.
I try to find a bit of time before the end of the week.
What do you think is better? Service, action, something other?

tcerqueira, 2022-01-11:
I think a service will do for now but either one them works for me :)

closed 2022-04-19 by github-actions[bot]