stereolabs/zed-ros-wrapper #736: Create PointCloud through Zed2 camera

ManChrys, 2021-06-21:
Hello guys!

Im trying to create pointcloud through Zed2 camera.

When im running $roslaunch zed_wrapper zed2.launch

the topics are the following:

geo@geo-X570-GAMING-X:~$ rostopic list
/clicked_point
/depthimage_to_laserscan/parameter_descriptions
/depthimage_to_laserscan/parameter_updates
/diagnostics
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/scan
/tf
/tf_static
/zed2/joint_states
/zed2/zed_node/atm_press
/zed2/zed_node/confidence/confidence_map
/zed2/zed_node/depth/camera_info
/zed2/zed_node/depth/depth_registered
/zed2/zed_node/depth/depth_registered/compressed
/zed2/zed_node/depth/depth_registered/compressed/parameter_descriptions
/zed2/zed_node/depth/depth_registered/compressed/parameter_updates
/zed2/zed_node/depth/depth_registered/compressedDepth
/zed2/zed_node/depth/depth_registered/compressedDepth/parameter_descriptions
/zed2/zed_node/depth/depth_registered/compressedDepth/parameter_updates
/zed2/zed_node/depth/depth_registered/theora
/zed2/zed_node/depth/depth_registered/theora/parameter_descriptions
/zed2/zed_node/depth/depth_registered/theora/parameter_updates
/zed2/zed_node/disparity/disparity_image
/zed2/zed_node/imu/data
/zed2/zed_node/imu/data_raw
/zed2/zed_node/imu/mag
/zed2/zed_node/left/camera_info
/zed2/zed_node/left/image_rect_color
/zed2/zed_node/left/image_rect_color/compressed
/zed2/zed_node/left/image_rect_color/compressed/parameter_descriptions
/zed2/zed_node/left/image_rect_color/compressed/parameter_updates
/zed2/zed_node/left/image_rect_color/compressedDepth
/zed2/zed_node/left/image_rect_color/compressedDepth/parameter_descriptions
/zed2/zed_node/left/image_rect_color/compressedDepth/parameter_updates
/zed2/zed_node/left/image_rect_color/theora
/zed2/zed_node/left/image_rect_color/theora/parameter_descriptions
/zed2/zed_node/left/image_rect_color/theora/parameter_updates
/zed2/zed_node/left/image_rect_gray
/zed2/zed_node/left/image_rect_gray/compressed
/zed2/zed_node/left/image_rect_gray/compressed/parameter_descriptions
/zed2/zed_node/left/image_rect_gray/compressed/parameter_updates
/zed2/zed_node/left/image_rect_gray/compressedDepth
/zed2/zed_node/left/image_rect_gray/compressedDepth/parameter_descriptions
/zed2/zed_node/left/image_rect_gray/compressedDepth/parameter_updates
/zed2/zed_node/left/image_rect_gray/theora
/zed2/zed_node/left/image_rect_gray/theora/parameter_descriptions
/zed2/zed_node/left/image_rect_gray/theora/parameter_updates
/zed2/zed_node/left_cam_imu_transform
/zed2/zed_node/left_raw/camera_info
/zed2/zed_node/left_raw/image_raw_color
/zed2/zed_node/left_raw/image_raw_color/compressed
/zed2/zed_node/left_raw/image_raw_color/compressed/parameter_descriptions
/zed2/zed_node/left_raw/image_raw_color/compressed/parameter_updates
/zed2/zed_node/left_raw/image_raw_color/compressedDepth
/zed2/zed_node/left_raw/image_raw_color/compressedDepth/parameter_descriptions
/zed2/zed_node/left_raw/image_raw_color/compressedDepth/parameter_updates
/zed2/zed_node/left_raw/image_raw_color/theora
/zed2/zed_node/left_raw/image_raw_color/theora/parameter_descriptions
[... 125 lines cut ...]
/zed2/zed_node/temperature/right

and when im opening rviz

[image]

i tried the solution which provided by @NeilNie on the issue  #393 
[file.txt](https://github.com/stereolabs/zed-ros-wrapper/files/6685753/file.txt)

but then i have the following errors

1)geo@geo-X570-GAMING-X:~$ rosrun cameratopointcloud cameratopointcloud 
[ INFO] [1624268972.931252525]: Node started
[ERROR] [1624268972.938962538]: "map" passed to lookupTransform argument target_frame does not exist. 
[ERROR] [1624268972.972319353]: "map" passed to lookupTransform argument target_frame does not exist. 
[ERROR] [1624268973.005662842]: "map" passed to lookupTransform argument target_frame does not exist. 
[ERROR] [1624268973.038983408]: "map" passed to lookupTransform argument target_frame does not exist. 
[ERROR] [1624268973.072320245]: "map" passed to lookupTransform argument target_frame does not exist. 
[ERROR] [1624268973.105651040]: "map" passed to lookupTransform argument target_frame does not exist. 
[ERROR] [1624268973.138989901]: "map" passed to lookupTransform argument target_frame does not exist. 

2)geo@geo-X570-GAMING-X:~/catkin_ws$ roslaunch zed_wrapper zed2.launch 
[ERROR] [1624268973.167275538]: Client [/point_cloud_transform] wants topic /zed2/zed_node/depth/depth_registered to have datatype/md5sum [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181], but our version has [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743]. Dropping connection.

Thanks in advance guys.

ManChrys, 2021-06-21:
I've also tried to create point cloud through the package depthimage_to_laserscan

$rosrun depthimage_to_laserscan depthimage_to_laserscan image:=/zed2/zed_node/depth/depth_registered

but again on LaderScanr component the Status is Error.

Thanks again.

Myzhar, 2021-06-21:
@ManChrys are you trying to generate a point cloud or a virtual laser scan? They are two different kinds of data.

ManChrys, 2021-06-21:
Hello @Myzhar 

i tried both. I can open a second issue for the virtual laser scan tho.

Myzhar, 2021-06-21:
Can you describe in detail the first problem?

ManChrys, 2021-06-21:
Well about the first problem.

i took the code from the #393 and i create a catkin_pkg for the cpp code, which name is cameratopointcloud and codes name is cameratopointcloud.cpp, so after i catkin_make ( no errors) and i tried to run the cpp. I change some things inside the code to match with the camera model (ZED2) and then i run the following commands:
1) $roslaunch zed_wrapper zed2.launch and 
2) $ rosrun cameratopointcloud cameratopointcloud 

in order to fix status error. After that i had the errors that i show above.

I understood something wrong ? Correct me plz so i can fix it :P

Thank you again.

Myzhar, 2021-06-22:
Sincerely I'm not sure of the goal that you are trying to reach. The code of issue #393 allows getting the point cloud in the map frame instead of the camera frame.
You are saying instead that you cannot see the point cloud in Rviz... 
I see two errors. The first, relative to the `map` frame is caused by the second: the ZED node is not correctly working so there is not a node publishing the correct TF tree, your catkin workspace seems broken...

Please try the following steps:
1) delete the `devel` and the `build` folders in your catkin workspace to clean the cache of CMake
2) `$ roscd zed_wrapper & git pull` to update the ZED ROS Wrapper
3) `$ roscd zed_examples & git pull` to update the ZED ROS Examples, if you are using them
4) go back into the root of your catkin workspace and `$ catkin_make` to recompile everything

Then try to run your code and let me know if this solved the problem

ManChrys, 2021-06-22:
ok i see what i misunderstanding here!!! 

Thanks for your time my friend!!

closed 2021-06-22 by ManChrys