stereolabs/zed-ros-wrapper #737: [Question]  Problem on the procedure of creation a virtual laser scanner through depth image (ZED2)

ManChrys, 2021-06-21:
Hello,

I've tried to create point cloud through the package depthimage_to_laserscan.

first i $roslaunch zed_wrapper zed2.launch

and then

$rosrun depthimage_to_laserscan depthimage_to_laserscan image:=/zed2/zed_node/depth/depth_registered

~~~
<launch>
<node name="depthimage_to_laserscan" pkg="depthimage_to_laserscan" type="depthimage_to_laserscan"> 
  <remap from="image" to="/zed2/zed_node/depth/depth_registered"/>
</node>
</launch>
~~~

but when i $rosrun rviz rviz

on LaserScan component the Status: Error.

[image]

Thanks in advance guys.

Myzhar, 2021-06-21:
Have you tried the example?
https://github.com/stereolabs/zed-ros-examples/tree/master/examples/zed_nodelet_example

ManChrys, 2021-06-22:
as always 100% correct and fast answers!!!

[image]

Ty @Myzhar  !! <3

closed 2021-06-22 by ManChrys