stereolabs/zed-ros-wrapper #738: [Question] How can I downgrade the SDK?

JoaoNBNeves, 2021-06-22:
I currently have the 3.4.2 SDK version and would like to downgrade it to 3.2.2.

Hou can I do this?

Thanks in advance.

Myzhar, 2021-06-22:
Hi @JoaoNBNeves 
if this question is not related to ROS please write an email to support@stereolabs.com or write a post on Stereolabs community website:
https://community.stereolabs.com/

However, downgrading is straightforward, simply start the ZED SDK installer, it will take care of cleaning the current installation, even if newer.

SimonSongg, 2021-06-22:
> Hi @JoaoNBNeves
> if this question is not related to ROS please write an email to [support@stereolabs.com](mailto:support@stereolabs.com) or write a post on Stereolabs community website:
> https://community.stereolabs.com/
> 
> However, downgrading is straightforward, simply start the ZED SDK installer, it will take care of cleaning the current installation, even if newer.

Hi @Myzhar , if I want to upgrade the SDK, the only thing I need to do is simply run the installer for the new version? Thanks.

Myzhar, 2021-06-22:
Hi @SimonSongg,
as I said to @JoaoNBNeves the SDK installer will take care of all the process, replacing the current installation with the new one.

closed 2021-06-22 by JoaoNBNeves