stereolabs/zed-ros-wrapper #742: [Question] Use ZED2 camera to measure true distance

liu978079474, 2021-06-28:
Hello, I am using the ZED2 camera. Using the SDK in the ROS system on Jetson NX, how do I convert 2D pixels into 3D space points?

Myzhar, 2021-06-28:
Hi @liu978079474 
in ROS you have all the data ready. The depth topic contains the map of the depths of all the pixels, will the point cloud topic contains the 3D coordinates of each of them.

liu978079474, 2021-06-28:
Hello, thank you for your reply. In the point cloud theme of ROS, how to associate 2D pixels with point cloud coordinates?

Myzhar, 2021-06-28:
The i-th pixel of the point cloud matched the same i-th pixel of the Left RGB image and the same i-th pixel of the depth map because all data are registered.

liu978079474, 2021-06-28:
Hello, thank you for your reply. Is there a way to calculate the point cloud corresponding to the 2D pixel through the depth value?

Myzhar, 2021-06-28:
Sure you can, but it's more complicated.
You must retrieve the camera pinhole model parameters from the depth map camera_info topic.
The depth map contains the depth, that is the "Z" coordinate (in camera frame).
You can calculate the X and Y coordinates as:
X = (u-c_x)*Z/f_x
X = (v-c_y)*Z/f_y
where (u,v) are the 2D coordinates, (c_x,c_y) are the optical center, (f_x,f_y) are the focal.

You can read more here:
https://docs.opencv.org/4.5.2/d9/d0c/group__calib3d.html

liu978079474, 2021-06-28:
Ok,thank you very much!

closed 2022-04-19 by github-actions[bot]