stereolabs/zed-ros-wrapper #744: [Question]

MarcoMaissan, 2021-06-30:
Hi! 
I'm using a ZED 2 with the zed_ros_wrapper on an Nvidia Jetson Xavier NX. I use a Yaml file for setting the parameters. When I try to set the `area_memory_db_path` to `~/test.area`, and `area_memory` to `true`, it throws the following error: `area_memory_db_path path doesn't exist or is unreachable.`

I tried to do `touch ~/test.area` to already create the file, but it returns the same error. The same happens when I use the absolute path (`/home/x/test.area`). Any idea why I keep getting this non-existing path error?

Myzhar, 2021-06-30:
Hi @MarcoMaissan 
the `area_memory_db_path` parameter requires an absolute path, so the first error is expected.
The second error is instead not expected. We will investigate the problem.

MarcoMaissan, 2021-06-30:
Thanks for the quick response! Also useful, when I use the absolute path and use the `touch test.area` file, I get the following: `Tracking not activated: INVALID ARE FILE`. I guess `ARE` is actually supposed to be `AREA`?

Myzhar, 2021-06-30:
Oh, yes. That's a typo. 
However, I figured out the problem. The ZED Wrapper cannot create a new area map, but it must use a map already created by another tool (see the positional tracking tutorial for an example: https://github.com/stereolabs/zed-examples/tree/master/positional%20tracking)
There is an old feature request about this problem:
https://github.com/stereolabs/zed-ros-wrapper/issues/547
we will add the command to start area map recording in the next few days.

dreamonJure, 2021-07-20:
Hi,  @Myzhar  @MarcoMaissan 
I also encountered a similar issue when I used `zed.saveAreaMap('my absolute path')`. The problem is that no `.area` file was generated by the usage of this function. And then I used `zed.getAreaExportState()` to check state, it showed 'file exported successfully'. I was just quite confused about it. So could you please give me some advice? Thank you very much.

KrasnovPavel, 2022-01-17:
Hi!
I encountered this issue too, but in zed-ros2-wrapper.

Myzhar, 2022-01-17:
There was an issue with the function `getAreaExportState` fixed with ZED SDK v3.6.3

closed 2022-04-19 by github-actions[bot]