stereolabs/zed-ros-wrapper #745: A problem about installing SDK of zed 2i

wildpig888, 2021-07-06:
Hi!
I met a error when installing the SDK 3.5.0。
my environment: cuda 10.2      ubuntu 18.04
[image]

The following packages have unsatisfied dependencies:
Libusb-1.0-0-dbg: dependency: libusb-1.0-0 (= 2:1.0.17-1ubuntu2), but 2:1.0.21-2 is about to be installed.

After that I install  libusb-1.0-0=2:1.0.17-1ubuntu2, but it still shows this error. I don't know how to solve it.
Thanks!

Myzhar, 2021-07-06:
@wildpig888 the correct library to install:
`sudo apt-get install libusb-1.0-0-dev`

wildpig888, 2021-07-06:
@Myzhar  thanks. I tried your advice but it shows 
[image]
zwx@zwxubuntu :~$ sudo apt-get install libusb-1.0-0-dev

[sudo] ZWX's password:

Reading package list... Done

Parsing dependency tree for package

Reading status information... Done

Libusb-1.0-0-dev is the latest version (2:1.0.21-2).

Libusb-1.0-0-dev is set to manual installation.

The following packages are installed automatically and are no longer required:

clang libbotan-2-4 libqbscore1.10 libqbsqtprofilesetup1.10

libqt5xmlpatterns5 libtspi1 linux-hwe-5.4-headers-5.4.0-42

linux-hwe-5.4-headers-5.4.0-48 linux-hwe-5.4-headers-5.4.0-51

linux-hwe-5.4-headers-5.4.0-52 linux-hwe-5.4-headers-5.4.0-53

linux-hwe-5.4-headers-5.4.0-62 linux-hwe-5.4-headers-5.4.0-66 qbs-common

qmlscene qtcreator-data qtcreator-doc qtdeclarative5-dev-tools

qtxmlpatterns5-dev-tools

Use 'sudo apt autoremove' to unload it (them).

0 software package has been upgraded and 0 software package has been newly installed. To uninstall 0 software package, 268 software packages have not been upgraded.

It shows Libusb-1.0-0-dev is the latest version (2:1.0.21-2). How can I change its version? I tried to remove but that will remove the desktop at the same time.

Myzhar, 2021-07-06:
The version is correct. Try to clean the catkin workspace deleting the `build` and `devel` folders, then start a new `catkin_make` process

wildpig888, 2021-07-06:
I have not installing zed-ros-wrapper. I met mistakes when I run the ZED SDK .
ZED_SDK_Ubuntu18_cuda10.2_v3.5.0.run
I didn't use catkin_make. I'm nearly mad.

Myzhar, 2021-07-06:
In this case, you're out of topic here. Please write an email to support@stereolabs.com to get direct support on your problem.

closed 2021-07-06 by Myzhar