stereolabs/zed-ros-wrapper #746: Zed2i ROS Node starts but don't see any topics?!?!?

swiz23, 2021-07-08:
Hi,

I installed sdk 3.5.0 using Cuda 11.1 on Ubuntu 18.04.5 (ROS Melodic). I then successfully compiled the zed_wrapper ROS package. Finally, I ran `roslaunch zed_wrapper zed2i.launch`

The node seems to start successfully - the only warning is the occasional low USB bandwidth. However, when I do a rostopic list, all i get are

```
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static
/zed2i/joint_states
```
..and the /zed2i/joint_states topic has no messages. Is there something I'm missing? Where are the image topics?

swiz23, 2021-07-08:
Just as an edit - it seems like the node starts and then after displaying *** Opening ZED 2i..., it hangs for a bit - then it displays ZED connection -> LOW USB BANDWIDTH. I'm using the data cable that came with the camera and am plugging it into a USB 3.0 port, so not sure what's going on...

swiz23, 2021-07-08:
Tried a different USB port and apparently that fixed the issue. All the ports on my laptop are USB-C, but I guess not all USB-C ports are created equal...

swiz23, 2021-07-08:
More on that. My setup is my camera connected to a USB 3.0 hub which connects to a USB - C port on my laptop. So when I connected the USB 3.0 hub to a USB 3.1 port on my laptop, I had issues. But when I connected the hub to a Thunderbolt 3 port, I had no issues. After doing a brief google, I found out that USB 3.1 can do speeds up to 10 Gb/s but Thunderbolt 3 ports can do up to 40 Gb/s. So I understand why the performance is better - but I don't see why I should even be having issues with my USB 3.1 port if the Zed 2i cable itself is USB 3.0. Unless  the hub introduces some weird delay?

Myzhar, 2021-07-08:
Hi @swiz23, is the USB3 hub powered?

swiz23, 2021-07-08:
No, it is not.

swiz23, 2021-07-08:
I'll see if I can get my hands on one. How much power (I imagine this boils down to current draw) does it need?

Myzhar, 2021-07-08:
The current required by the camera itself is not high, 500 mA can be enough, but if you have other devices connected to the hub and the current sourced by the port is limited, then the camera and the other devices can fail to bootstrap.

swiz23, 2021-07-08:
I imagine that all usb ports on a laptop can provide at least that much current, and I don't have anything else plugged into the hub.

As an update, I flipped the USB-C connector on my hub when plugging into the non-thunderbolt port on my laptop, and that works! I'm not sure why it does, but I've tested using the connector flipped vs not flipped multiple times, and it works every time when I flip it, and only occasionally if I don't flip it.

Could be a hub issue or something up with the port on my laptop. Either way - it doesn't seem like this is related to the Zed 2i camera so I'm closing the issue.

closed 2021-07-08 by swiz23