stereolabs/zed-ros-wrapper #747: obstacle detection module not responding after some time

dineshrboson, 2021-07-09:
Zed obstacle detection module is not working after launching and running for some time . let me know how to resume it or how to avoid it.
[image]

Myzhar, 2021-07-09:
@dineshrboson you are not using the latest version of the ZED Wrapper or you have not cleaned the CMake cache after an update.
Please delete the `build` and `devel` sources and then update `zed-ros-wrapper` and `zed-ros-examples` (if you are using it):
```
$ roscd zed_wrapper
$ git pull
$ roscd zed_examples
$ git pull
$ roscd
$ cd ..
$ rm -rf devel
$ rm -rf build
$ catkin_make
```

dineshrboson, 2021-07-09:
everything is latest version and done for the first time

dineshrboson, 2021-07-09:
I think there is problem with zed obstacle detection module please check once and I am pretty sure about it because in past I found the same issue with another zed camera

Myzhar, 2021-07-09:
What is the ZED Obstacle Detection module? We do not have a repository with a similar name

Myzhar, 2021-07-09:
Supposing that you are referring to the `zed_obj_det_sub_tutorial` I can confirm that it works as expected:
```
 $ rosrun zed_obj_det_sub_tutorial zed_obj_det_sub objects:=/zed2i/zed_node/obj_det/objects
[ INFO] [1625833895.008642555]: ***** New object list *****
[ INFO] [1625833895.009372719]: Person [0] - Pos. [0.897203,-0.485077,-0.217374] [m]- Conf. 86.743 - Tracking state: 1
[ INFO] [1625833895.042787580]: ***** New object list *****
[ INFO] [1625833895.042822278]: Person [0] - Pos. [0.897253,-0.485541,-0.217522] [m]- Conf. 86.9598 - Tracking state: 1
[ INFO] [1625833895.075379118]: ***** New object list *****
[ INFO] [1625833895.075416392]: Person [0] - Pos. [0.898212,-0.485767,-0.217622] [m]- Conf. 86.7255 - Tracking state: 1
```
indeed the topic is correctly published:
```
$ rostopic echo /zed2i/zed_node/obj_det/objects
header: 
  seq: 490
  stamp: 
    secs: 1625834011
    nsecs: 261494594
  frame_id: "zed2i_left_camera_frame"
objects: 
  - 
    label: "Person"
    label_id: 0
    sublabel: "Person"
    confidence: 86.29715728759766
    position: [0.8717639446258545, -0.47807252407073975, -0.21728461980819702]
    position_covariance: [0.0003998384636361152, -0.00021927006309852004, -9.965855861082673e-05, 0.00012778572272509336, 5.4652417020406574e-05, 3.4380842407699674e-05]
    velocity: [0.008411629125475883, -0.0009302149992436171, 0.006421429570764303]
    tracking_available: True
    tracking_state: 1
    action_state: 0
    bounding_box_2d: 
      corners: 
        - 
          kp: [690, 243]
        - 
          kp: [1023, 243]
        - 
          kp: [1023, 717]
        - 
          kp: [690, 717]
    bounding_box_3d: 
      corners: 
        - 
          kp: [0.41472625732421875, -0.19629324972629547, -0.08309848606586456]
        - 
          kp: [0.8584190607070923, -0.18884168565273285, 0.25436824560165405]
        - 
          kp: [0.8584190607070923, -0.7462025880813599, 0.26667529344558716]
        - 
          kp: [0.41472625732421875, -0.7536542415618896, -0.07079143822193146]
        - 
          kp: [0.8851088285446167, -0.20994248986244202, -0.7012445330619812]
        - 
          kp: [1.3288016319274902, -0.2024909108877182, -0.3637778162956238]
        - 
          kp: [1.3288016319274902, -0.7598518133163452, -0.3514707684516907]
        - 
          kp: [0.8851088285446167, -0.767303466796875, -0.6889374852180481]
    dimensions_3d: [0.5574968457221985, 0.7768852114677429, 0.5574967265129089]
    skeleton_available: True
    head_bounding_box_2d: 
      corners: 
        - 
          kp: [1.1658803223182478e-42, 3.60133705331478e-43]
        - 
          kp: [1.593276353937317e-42, 3.60133705331478e-43]
        - 
          kp: [1.593276353937317e-42, 6.151700258385947e-43]
        - 
[... 92 lines cut ...]
        - 
          kp: [nan, nan, nan]
        - 
          kp: [0.7038967609405518, -0.3780359625816345, -0.04458138346672058]
---
header: 
  seq: 491
  stamp: 
    secs: 1625834011
    nsecs: 294673594
  frame_id: "zed2i_left_camera_frame"
objects: 
  - 
    label: "Person"
    label_id: 0
    sublabel: "Person"
    confidence: 86.31736755371094
    position: [0.8749319314956665, -0.4794239103794098, -0.21797597408294678]
    position_covariance: [0.00040568222175352275, -0.0002222958573838696, -0.00010106956324307248, 0.00012940190208610147, 5.538163532037288e-05, 3.4790660720318556e-05]
    velocity: [0.007215602323412895, -0.00036069349152967334, 0.005498555488884449]
    tracking_available: True
    tracking_state: 1
    action_state: 0
    bounding_box_2d: 
```

dineshrboson, 2021-07-13:
[image]
[image]
[image]
[image]

dineshrboson, 2021-07-13:
After running the display zed2 launch file which has object detection module enabled for 1 hour 
detection stopped to work again.

dineshrboson, 2021-07-13:
i tried to enable it again with service call but it didn't worked

Myzhar, 2021-07-13:
Can you please explain better your configuration in detail?
It seems like you are performing something that invalidates your environment settings.
You can also try to use `rqt` to start the service being sure to use the correct parameters instead of writing them manually

closed 2022-04-19 by github-actions[bot]