stereolabs/zed-ros-wrapper #748: [Question] No data published to any topic, ZED2 in Docker

XDGFX, 2021-07-13:
I'm having trouble setting up a ZED 2 camera.

Running on a Jetson Xavier NX (Ubuntu 18.04, L4T 32.5.0). I've set up a Docker container for the 3.5 SDK and wrapper (although for reference, the same issue is happening with the provided `3.4-ros-devel-jetson-jp4.3` image).

The ROS master is running on the host, and the container is using bridge networking. Therefore, I set the environment variable `ROS_MASTER_URI=http://172.17.0.1:11311`. I can start the wrapper using `roslaunch zed_wrapper zed2.launch`, and topics appear on the host as expected.

However I cannot get any data from the topics, they all appear empty. I've tried opening rviz with the zed2 config provided in the examples repo. I've also tried echoing various topics, and using `image_view` to show a raw image (`rosrun image_view image_view image:/zed2/zed_node/right/image_raw_color`).

Hopefully this is just a simple config issue, help is much appreciated!

For reference, the startup appears fine, shown below:

```log
root@b5fffc1e3fc1:/opt/ros_ws# roslaunch zed_wrapper zed2.launch 
... logging to /root/.ros/log/bb1844bc-e3be-11eb-a859-48b02d0788d9/roslaunch-b5fffc1e3fc1-187.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://b5fffc1e3fc1:45665/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /zed2/zed2_description: <?xml version="1....
 * /zed2/zed_node/auto_exposure_gain: True
 * /zed2/zed_node/auto_whitebalance: True
 * /zed2/zed_node/brightness: 4
 * /zed2/zed_node/contrast: 4
 * /zed2/zed_node/depth/depth_downsample_factor: 1.0
 * /zed2/zed_node/depth/depth_stabilization: 1
 * /zed2/zed_node/depth/max_depth: 20.0
 * /zed2/zed_node/depth/min_depth: 0.7
 * /zed2/zed_node/depth/openni_depth_mode: False
 * /zed2/zed_node/depth/quality: 1
 * /zed2/zed_node/depth/sensing_mode: 0
 * /zed2/zed_node/depth_confidence: 50
 * /zed2/zed_node/depth_texture_conf: 100
 * /zed2/zed_node/exposure: 100
 * /zed2/zed_node/gain: 100
 * /zed2/zed_node/gamma: 8
 * /zed2/zed_node/general/base_frame: base_link
 * /zed2/zed_node/general/camera_flip: False
 * /zed2/zed_node/general/camera_model: zed2
 * /zed2/zed_node/general/camera_name: zed2
 * /zed2/zed_node/general/gpu_id: -1
 * /zed2/zed_node/general/grab_frame_rate: 30
 * /zed2/zed_node/general/resolution: 2
 * /zed2/zed_node/general/self_calib: True
 * /zed2/zed_node/general/serial_number: 0
 * /zed2/zed_node/general/svo_compression: 2
 * /zed2/zed_node/general/verbose: False
 * /zed2/zed_node/general/zed_id: 0
 * /zed2/zed_node/hue: 0
 * /zed2/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zed2/zed_node/mapping/mapping_enabled: False
 * /zed2/zed_node/mapping/max_mapping_range: -1
 * /zed2/zed_node/mapping/resolution: 0.05
 * /zed2/zed_node/object_detection/confidence_threshold: 50
 * /zed2/zed_node/object_detection/object_tracking_enabled: True
 * /zed2/zed_node/object_detection/od_enabled: False
 * /zed2/zed_node/object_detection/people_detection: True
 * /zed2/zed_node/object_detection/vehicle_detection: False
 * /zed2/zed_node/point_cloud_freq: 15.0
 * /zed2/zed_node/pos_tracking/area_memory: True
 * /zed2/zed_node/pos_tracking/area_memory_db_path: 
 * /zed2/zed_node/pos_tracking/fixed_z_value: 0.0
[... 134 lines cut ...]
[ INFO] [1626170264.837172368]: Advertised on topic /zed2/zed_node/right/image_rect_gray
[ INFO] [1626170264.837587191]: Advertised on topic /zed2/zed_node/right/camera_info
[ INFO] [1626170264.969722850]: Advertised on topic /zed2/zed_node/right_raw/image_raw_gray
[ INFO] [1626170264.969932390]: Advertised on topic /zed2/zed_node/right_raw/camera_info
[ INFO] [1626170265.095583522]: Advertised on topic /zed2/zed_node/depth/depth_registered
[ INFO] [1626170265.095822150]: Advertised on topic /zed2/zed_node/depth/camera_info
[ INFO] [1626170265.219021144]: Advertised on topic /zed2/zed_node/stereo/image_rect_color
[ INFO] [1626170265.341065046]: Advertised on topic /zed2/zed_node/stereo_raw/image_raw_color
[ INFO] [1626170265.344239724]: Advertised on topic /zed2/zed_node/confidence/confidence_map
[ INFO] [1626170265.348086830]: Advertised on topic /zed2/zed_node/disparity/disparity_image
[ INFO] [1626170265.352174963]: Advertised on topic /zed2/zed_node/point_cloud/cloud_registered
[ INFO] [1626170265.356636799]: Advertised on topic /zed2/zed_node/pose
[ INFO] [1626170265.360876872]: Advertised on topic /zed2/zed_node/pose_with_covariance
[ INFO] [1626170265.364338562]: Advertised on topic /zed2/zed_node/odom
[ INFO] [1626170265.368295590]: Advertised on topic /zed2/zed_node/path_odom
[ INFO] [1626170265.372292170]: Advertised on topic /zed2/zed_node/path_map
[ INFO] [1626170265.375921352]: Advertised on topic /zed2/zed_node/imu/data
[ INFO] [1626170265.378917851]: Advertised on topic /zed2/zed_node/imu/data_raw
[ INFO] [1626170265.382444471]: Advertised on topic /zed2/zed_node/imu/mag
[ INFO] [1626170265.386090325]: Advertised on topic /zed2/zed_node/temperature/imu
[ INFO] [1626170265.389610801]: Advertised on topic /zed2/zed_node/atm_press
[ INFO] [1626170265.392938538]: Advertised on topic /zed2/zed_node/temperature/left
[ INFO] [1626170265.396099391]: Advertised on topic /zed2/zed_node/temperature/right
[ INFO] [1626170265.399623931]: Advertised on topic /zed2/zed_node/left_cam_imu_transform [LATCHED]
```

Myzhar, 2021-07-13:
Hi @XDGFX 
are the topics empty inside or outside the container?

Myzhar, 2021-07-13:
Check that all the ports used by ROS are opened:
http://wiki.ros.org/Security

XDGFX, 2021-07-13:
Got it working 👍.

As ROS1 seems to need a lot of ports open I've set the networking to host (`network_mode: host` in docker-compose, alternatively the flag `--net=host`), and that seems to have fixed it.

Many thanks for the nearly instant help :)

closed 2021-07-13 by XDGFX

Leovilhena, 2021-09-07:
@XDGFX would it be possible for you to share the Dockerfile for ZED2?
I've been struggling to write a working one and any help would be really appreciated.

Cheers

XDGFX, 2021-09-07:
@Leovilhena The Dockerfile I'm using is here :smile: 
https://github.com/stereolabs/zed-ros2-wrapper/issues/48#issuecomment-892700306