stereolabs/zed-ros-wrapper #749: [Question] Is it possible to save a pointcloud using the API (maybe via a service)?

flabrosse, 2021-07-14:
I'm looking into using the mapping feature of the ZED API (with a ZED camera) to create a point cloud using ROS while my robot drives around.  I need to save this point cloud at the end.  It seems that the only way to do this is to subscribe to mapping/fused_cloud and save the point cloud that is received.  I don't need the intermediate clouds so would either only need the last one or, better, to tell the ZED API to save the current cloud.  Is this possible using the ROS wrapper?  I did not see anything that looks like it would allow it.

Thanks

Myzhar, 2021-07-14:
Hi @flabrosse 
there is not currently a feature that allows saving the current fused point cloud.
I added the "Feature request" tag to this issue to remember to add a service to add the feature to a future release.

flabrosse, 2021-07-14:
Hi @Myzhar, thank you for the quick response.  OK, for now I'll have to do with subscribing.

closed 2021-07-27 by Myzhar

flabrosse, 2021-08-02:
Thank you very much for doing this!  I really appreciate the quick responses and improvements!

So this saves locally (in my case on the jetson running the ZED software).  Could we also have a service to get the pointcloud on a different computer?  Maybe I should open a new feature request.

Myzhar, 2021-08-02:
In this case you must use an external node like `pointcloud_to_pcd`:
https://wiki.ros.org/pcl_ros#pointcloud_to_pcd

flabrosse, 2021-08-02:
Although PCD is the wrong format for me (because of limitations of precision in saved numbers).  I now have code that saves a pointcloud to LAS, but I need to receive the pointcloud, and ideally only at the end, not every so often.  But I will look at saving using the new service and using ftp (or some such to get it locally).