stereolabs/zed-ros-wrapper #750: [Feature] zed_interfaces as a package?

flabrosse, 2021-07-15:
I'm trying to use services to talk to the ZED from a different computer than where the ZED is connected (and where the zed.launch is started).  This implies having zed_interfaces on this computer.  As far as I can see, the only way to have that is to install the whole of the ros wrapper.

Could we have a separate package for zed_interfaces?

This is made even more needed since on the latest ubuntu (20.04), the install fails because the package python-geographiclib is now python3-geographiclib.

Cheers,
Fred

Myzhar, 2021-07-15:
@flabrosse `zed_interfaces` is already on a separate package, it's in the same Github repository, but the ROS packages are all separated. 
You can simply copy the folder `zed_interfaces` in the `src` folder of the catkin workspace of the second computer and start to compile to make everything work.

flabrosse, 2021-07-15:
Hi @Myzhar, thanks again for the quick response.  Yes, this is what I ended up doing, but this also means that if I want to do a git pull, the whole thing will be pulled again.  Not the end of the world, just annoying.

Thanks

davesarmoury, 2021-07-21:
+1 for this request. Having interfaces and description packages in their own repo would make a distributed application easier. This repo can even include the new repo as a git sub module

Myzhar, 2021-07-21:
I see what I can do about this request 👍

Myzhar, 2021-07-26:
The `zed_interfaces` package has been removed from this repository and moved to its own [`zed-ros-interfaces` repository](https://github.com/stereolabs/zed-ros-interfaces) for allowing better integration of the ZED Wrapper on remote ground stations that do not require the full package to be installed. To update your repository please follow the [new update instructions](https://github.com/stereolabs/zed-ros-wrapper#update-the-repository).

closed 2021-07-26 by Myzhar

davesarmoury, 2021-07-26:
@Myzhar The meshes from https://github.com/stereolabs/zed-ros-wrapper/tree/master/zed_wrapper/urdf/models should also be moved so you can use rviz remotely without getting errors

Myzhar, 2021-07-26:
Yes, it makes sense because they are loaded by Rviz and not propagated with the "robot description".

Myzhar, 2021-07-26:
@davesarmoury done. Now the meshes are in the `zed_interfaces` repository and they can be correctly loaded by Rviz

davesarmoury, 2021-07-26:
Wonderful! Small improvement that will make life much easier :)

Myzhar, 2021-07-26:
Thank you for your feedback

flabrosse, 2021-08-02:
Thank you indeed!

flabrosse, 2021-08-04:
Any particular reason why the package is called zed-ros-interfaces rather than zed_ros_interfaces?  The latter is the ROS convention (I myself are in the process of renaming things...)

Myzhar, 2021-08-04:
Because that's not the name of the package, but the name of the Github repository.
The names of the ROS packages included in this repository follow the ROS naming standard:
`zed_ros` -> the meta-package
`zed_wrapper` -> the package that contains the ZED Wrapper
`zed_ros_interfaces` -> the sub-module including the definitions of the custom topics and services
`zed_nodelet` -> the package that contains the ZED Nodelets

flabrosse, 2021-08-04:
Fair enough.  I always find it confusing that the top directory is not called the same as the package, but I have seen this before when there is more than one package.