stereolabs/zed-ros-wrapper #753: [Question] how to change image publish rate？

zhaohaixu, 2021-07-16:
I want to change the publish rate of image_raw_color topic from 15HZ to 30 HZ, can any one tell me how to change pub_frame_rate in code zed_wrapper_nodelet.cpp? Thanks ! :)

Myzhar, 2021-07-16:
Hi @zhaohaixu 
the `pub_frame_rate` is a parameter of the node, you can change it simply by modifying the [`common.yaml`](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L19) file or by modifying it dynamically while the node is running.
Please remember that the parameter [`general/grab_frame_rate`](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L27) must be changed to 30 (or 60 or 100) otherwise the effective framerate will be reduced.

zhaohaixu, 2021-07-16:
> Hi @zhaohaixu
> the `pub_frame_rate` is a parameter of the node, you can change it simply by modifying the [...]

Got it, it works well! Thanks!

closed 2021-07-16 by Myzhar