stereolabs/zed-ros-wrapper #757: [BUG]When I roslaunch zed_wrapper zed2.launch ,tf_tree is not correct and the MAP ODOM IMU is missing

lnikki, 2021-07-23:
**Describe the bug**
A clear and concise description of what the bug is.

**Device information:**
 
 - ZED SDK Version: v3.5.0
 - ROS Wrapper version: [latest master]

When I roslaunch zed_wrapper zed2.launch ,tf_tree is not correct and the MAP ODOM IMU is missing.There is no output on the topic of imu_data.

**Screenshots**
[image]

**Additional context**
comman.yaml
pos_tracking:
    pos_tracking_enabled:       true                            # True to enable positional tracking from start
    publish_tf:                 true                            # publish `odom -> base_link` TF
    publish_map_tf:             true                            # publish `map -> odom` TF
    map_frame:                  'map'
    odometry_frame:             'odom'
    area_memory_db_path:        ''
    area_memory:                true                            # Enable to detect loop closure
    floor_alignment:            false                           # Enable to automatically calculate camera/floor offset
    initial_base_pose:          [0.0,0.0,0.0, 0.0,0.0,0.0]      # Initial position of the `base_frame` -> [X, Y, Z, R, P, Y]
    init_odom_with_first_valid_pose: true                       # Enable to initialize the odometry with the first valid pose
    path_pub_rate:              2.0                             # Camera trajectory publishing frequency
    path_max_count:             -1                              # use '-1' for unlimited path size
    two_d_mode:                 false                           # Force navigation on a plane. If true the Z value will be fixed to "fixed_z_value", roll and pitch to zero
    fixed_z_value:              0.00                            # Value to be used for Z coordinate if `two_d_mode` is true
  
zed2.yaml
sensors:
    sensors_timestamp_sync:     false           # Synchronize Sensors messages timestamp with latest received frame
    publish_imu_tf:             true            # publish `IMU -> <cam_name>_left_camera_frame` TF

Myzhar, 2021-07-26:
Hi @lnikki this is strange.
Can you post the console log?

lnikki, 2021-07-28:
WARNING: Package name "ZED_Object_detection_image_viewer" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "ZED_Spatial_Mapping" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "ZED_Depth_Sensing" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "ZED_Point_Cloud_Mapping" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "ZED_Camera_Control" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "ZED_Body_Tracking" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
... logging to /home/nuannuan/.ros/log/27901b84-ef53-11eb-838b-e99943524033/roslaunch-nuannuan-Lenovo-XiaoXinPro-13IML-2020-11493.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "ZED_Object_detection_image_viewer" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "ZED_Spatial_Mapping" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "ZED_Depth_Sensing" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "ZED_Point_Cloud_Mapping" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "ZED_Camera_Control" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "ZED_Body_Tracking" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
started roslaunch server http://nuannuan-lenovo-xiaoxinpro-13iml-2020:35939/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.8
 * /zed2/zed2_description: <?xml version="1....
 * /zed2/zed_node/auto_exposure_gain: True
 * /zed2/zed_node/auto_whitebalance: True
 * /zed2/zed_node/brightness: 4
 * /zed2/zed_node/contrast: 4
 * /zed2/zed_node/depth/depth_downsample_factor: 1.0
 * /zed2/zed_node/depth/depth_stabilization: 1
 * /zed2/zed_node/depth/max_depth: 20.0
 * /zed2/zed_node/depth/min_depth: 0.3
 * /zed2/zed_node/depth/openni_depth_mode: False
 * /zed2/zed_node/depth/quality: 1
 * /zed2/zed_node/depth/sensing_mode: 0
 * /zed2/zed_node/depth_confidence: 50
 * /zed2/zed_node/depth_texture_conf: 100
 * /zed2/zed_node/exposure: 100
 * /zed2/zed_node/gain: 100
 * /zed2/zed_node/gamma: 8
 * /zed2/zed_node/general/base_frame: base_link
 * /zed2/zed_node/general/camera_flip: False
 * /zed2/zed_node/general/camera_model: zed2
 * /zed2/zed_node/general/camera_name: zed2
 * /zed2/zed_node/general/gpu_id: -1
 * /zed2/zed_node/general/grab_frame_rate: 30
 * /zed2/zed_node/general/resolution: 2
 * /zed2/zed_node/general/self_calib: True
 * /zed2/zed_node/general/serial_number: 0
 * /zed2/zed_node/general/svo_compression: 2
 * /zed2/zed_node/general/verbose: False
 * /zed2/zed_node/general/zed_id: 0
 * /zed2/zed_node/hue: 0
 * /zed2/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zed2/zed_node/mapping/mapping_enabled: False
 * /zed2/zed_node/mapping/max_mapping_range: -1
 * /zed2/zed_node/mapping/resolution: 0.05
 * /zed2/zed_node/object_detection/body_fitting: False
 * /zed2/zed_node/object_detection/confidence_threshold: 50
 * /zed2/zed_node/object_detection/max_range: 15.0
 * /zed2/zed_node/object_detection/mc_animal: True
 * /zed2/zed_node/object_detection/mc_bag: True
 * /zed2/zed_node/object_detection/mc_electronics: True
 * /zed2/zed_node/object_detection/mc_fruit_vegetable: True
 * /zed2/zed_node/object_detection/mc_people: True
 * /zed2/zed_node/object_detection/mc_vehicle: True
 * /zed2/zed_node/object_detection/model: 3
 * /zed2/zed_node/object_detection/object_tracking_enabled: True
[... 174 lines cut ...]
[ INFO] [1627442606.831932363]: Advertised on topic /zed2/zed_node/path_map
[ INFO] [1627442606.833322849]: Advertised on topic /zed2/zed_node/imu/data
[ INFO] [1627442606.834552623]: Advertised on topic /zed2/zed_node/imu/data_raw
[ INFO] [1627442606.835804933]: Advertised on topic /zed2/zed_node/imu/mag
[ INFO] [1627442606.836916895]: Advertised on topic /zed2/zed_node/temperature/imu
[ INFO] [1627442606.838121378]: Advertised on topic /zed2/zed_node/atm_press
[ INFO] [1627442606.839307839]: Advertised on topic /zed2/zed_node/temperature/left
[ INFO] [1627442606.840457670]: Advertised on topic /zed2/zed_node/temperature/right
[... cut ...]
[ INFO] [1627442606.857070007]:  * Rotation: {0.000,-0.000,0.000}
[ INFO] [1627442606.857147470]: Static transform Camera Center to Base [zed2_camera_center -> base_link]
[ INFO] [1627442606.857281596]:  * Translation: {0.000,0.000,0.000}
[ INFO] [1627442606.857355520]:  * Rotation: {0.000,-0.000,0.000}
[ INFO] [1627442606.957579189]: Initial ZED left camera pose (ZED pos. tracking): 
[ INFO] [1627442606.957792036]:  * T: [0,0.06,0]
[ INFO] [1627442606.957925938]:  * Q: [0,0,0,1]
[ INFO] [1627442607.301221548]: Odometry aligned to last tracking pose
@Myzhar

Myzhar, 2021-07-28:
The log of the ZED Wrapper is correct. Can you run the `$ roswtf` command and post the log to see if there is something wrong at ROS level?

lnikki, 2021-07-28:
Static checks summary:

Found 1 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING ROS_HOSTNAME may be incorrect: ROS_HOSTNAME [nuannuan-lenovo-xiaoxinpro-13iml-2020] resolves to [127.0.1.1], which does not appear to be a local IP address ['127.0.0.1'].

================================================================================

ROS Master does not appear to be running.
Online graph checks will not be run.
ROS_MASTER_URI is [http://nuannuan-lenovo-xiaoxinpro-13iml-2020:11311]
@Myzhar

Myzhar, 2021-07-28:
Are the nodes running? You must call `roswtf` in a new console when all the nodes are running

lnikki, 2021-07-29:
Static checks summary:

Found 1 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING ROS_HOSTNAME may be incorrect: ROS_HOSTNAME [nuannuan-lenovo-xiaoxinpro-13iml-2020] resolves to [127.0.1.1], which does not appear to be a local IP address ['127.0.0.1', '192.168.43.217'].

================================================================================
Beginning tests of your ROS graph. These may take a while...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules
running tf checks, this will take a second...
... tf checks complete

Online checks summary:

Found 2 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING The following node subscriptions are unconnected:
 * /zed2/zed2_state_publisher:
   * /zed2/joint_states

WARNING No tf messages

@Myzhar

Myzhar, 2021-07-29:
Please run also `$ rqt_graph` and post the graph that it generates

lnikki, 2021-07-30:
[image]
@Myzhar

Myzhar, 2021-07-30:
The nodes graph is correct too. IS it possible that you changed any parameters breaking up the TF chain?
Have you tried changing the `base_frame`, the `map_frame`, or the `odometry_frame` parameters?

Can you please attach the common.yaml and zed2.yaml file?

lnikki, 2021-07-30:
I didn't change any of the  parameters @Myzhar 
common.yaml
# params/common.yaml
# Common parameters to Stereolabs ZED and ZED mini cameras
---

# Dynamic parameters cannot have a namespace
brightness:                 4                                   # Dynamic
contrast:                   4                                   # Dynamic
hue:                        0                                   # Dynamic
saturation:                 4                                   # Dynamic
sharpness:                  4                                   # Dynamic
gamma:                      8                                   # Dynamic - Requires SDK >=v3.1
auto_exposure_gain:         true                                # Dynamic
gain:                       100                                 # Dynamic - works only if `auto_exposure_gain` is false
exposure:                   100                                 # Dynamic - works only if `auto_exposure_gain` is false
auto_whitebalance:          true                                # Dynamic
whitebalance_temperature:   42                                  # Dynamic - works only if `auto_whitebalance` is false
depth_confidence:           50                                  # Dynamic
depth_texture_conf:         100                                 # Dynamic
pub_frame_rate:             15.0                                # Dynamic - frequency of publishing of video and depth data
point_cloud_freq:           15.0                                # Dynamic - frequency of the pointcloud publishing (equal or less to `grab_frame_rate` value)

general:
    camera_name:                zed                             # A name for the camera (can be different from camera model and node name and can be overwritten by the launch file)
    zed_id:                     0
    serial_number:              0
    resolution:                 2                               # '0': HD2K, '1': HD1080, '2': HD720, '3': VGA
    grab_frame_rate:            30                              # Frequency of frame grabbing for internal SDK operations
    gpu_id:                     -1
    base_frame:                 'base_link'                     # must be equal to the frame_id used in the URDF file
    verbose:                    false                           # Enable info message by the ZED SDK
    svo_compression:            2                               # `0`: LOSSLESS, `1`: AVCHD, `2`: HEVC
    self_calib:                 true                            # enable/disable self calibration at starting
    camera_flip:                false

video:
    img_downsample_factor:      1.0                             # Resample factor for images [0.01,1.0] The SDK works with native image sizes, but publishes rescaled image.
    extrinsic_in_camera_frame:  true                            # if `false` extrinsic parameter in `camera_info` will use ROS native frame (X FORWARD, Z UP) instead of the camera frame (Z FORWARD, Y DOWN) [`true` use old behavior as for version < v3.1]

depth:
    quality:                    1                               # '0': NONE, '1': PERFORMANCE, '2': QUALITY, '3': ULTRA
    sensing_mode:               0                               # '0': STANDARD, '1': FILL (not use FILL for robotic applications)
    depth_stabilization:        1                               # `0`: disabled, `1`: enabled
    openni_depth_mode:          false                           # 'false': 32bit float meters, 'true': 16bit uchar millimeters
    depth_downsample_factor:    1.0                             # Resample factor for depth data matrices [0.01,1.0] The SDK works with native data sizes, but publishes rescaled matrices (depth map, point cloud, ...)

pos_tracking:
    pos_tracking_enabled:       true                            # True to enable positional tracking from start
    publish_tf:                 true                            # publish `odom -> base_link` TF
    publish_map_tf:             true                            # publish `map -> odom` TF
    map_frame:                  'map'
    odometry_frame:             'odom'
    area_memory_db_path:        ''
    area_memory:                true                            # Enable to detect loop closure
    floor_alignment:            false                           # Enable to automatically calculate camera/floor offset
    initial_base_pose:          [0.0,0.0,0.0, 0.0,0.0,0.0]      # Initial position of the `base_frame` -> [X, Y, Z, R, P, Y]
    init_odom_with_first_valid_pose: true                       # Enable to initialize the odometry with the first valid pose
    path_pub_rate:              2.0                             # Camera trajectory publishing frequency
    path_max_count:             -1                              # use '-1' for unlimited path size
    two_d_mode:                 false                           # Force navigation on a plane. If true the Z value will be fixed to "fixed_z_value", roll and pitch to zero
    fixed_z_value:              0.00                            # Value to be used for Z coordinate if `two_d_mode` is true

mapping:
    mapping_enabled:            false                           # True to enable mapping and fused point cloud pubblication
    resolution:                 0.05                            # maps resolution in meters [0.01f, 0.2f]
    max_mapping_range:          -1                              # maximum depth range while mapping in meters (-1 for automatic calculation) [2.0, 20.0]
    fused_pointcloud_freq:      1.0                             # frequency of the publishing of the fused colored point cloud

zed2.yaml
[image]

Myzhar, 2021-07-30:
At this point, the only problem that I notice with your system is the warning about the IP address that `$ roswtf` writes out:
```
WARNING ROS_HOSTNAME may be incorrect: ROS_HOSTNAME [nuannuan-lenovo-xiaoxinpro-13iml-2020] resolves to [127.0.1.1], which does not appear to be a local IP address ['127.0.0.1'].
```
Please check it.
Furthermore, be sure that no other node is publishing the `odom` TF that can be in conflict with the ZED odom TF.

lnikki, 2021-07-30:
I corrected the wranning，but the problem still hasn't been resolved.
When I only $roslaunch zed_wrapper zed2.launch , $ roswtf writes out:
[image]
@Myzhar

Myzhar, 2021-07-31:
I have no more ideas about what can generate your problem. The ZED ROS Wrapper is well tested and following the installation guide you should get a working system, try to delete the `<catkin_ws>/src/zed-ros-wrapper` and the `<catkin_ws>/src/zed-ros-examples` folders and reinstall both from scratch, following the instructions of the README file.
Please be sure to use the latest version of the ZED SDK to exclude any possible compatibility problem.

lnikki, 2021-08-04:
Thanks for your advice, but I downloaded the SDK 3.5.1 and it still doesn't work. Could you please tell me the program for publishing the imu topic separately? @Myzhar

Myzhar, 2021-08-04:
There is not a program to publish the IMU topic separately. What do you mean with it? Maybe I misunderstood the question...

lnikki, 2021-08-04:
There is no data for the IMU topic I published using zed-ros-wrapper, so I would like to ask if there is a C file for publishing the IMU topic. I will check the system again.Thanks! @Myzhar

Myzhar, 2021-08-05:
@lnikki are you sure that your camera is working as expected.

As you can see in the following screenshot everything works as expected using:
`$ roslaunch zed_wrapper zed2.launch`
and
`$ rostopic echo /zed2/zed_node/imu/data`

[image]

Can you please run the `ZED_Diagnostic` command and upload the report it generates?

lnikki, 2021-08-06:
[ZED_Diagnostic_Results.txt](https://github.com/stereolabs/zed-ros-wrapper/files/6944298/ZED_Diagnostic_Results.txt)
@Myzhar

Myzhar, 2021-08-06:
@lnikki the camera is correctly detected and the OS is correctly configured.
The GPU is not really powerful, but it is good enough to correctly run the ZED node with the default parameters.

Have you tried on another PC if you have the same problems?

lnikki, 2021-08-09:
My friend's computer works, but I don't know what's wrong with mine. @Myzhar

closed 2022-04-19 by github-actions[bot]