stereolabs/zed-ros-wrapper #761: [BUG] zed_nodelets: Cannot locate rosdep definition for [zed_interfaces]

joseandres15, 2021-07-26:
**Describe the bug**
Hello,
Today, I decided to remove the catkin_ws and build it again. I was following the steps from (https://www.stereolabs.com/docs/ros/) to install zed-ros-wrapper again. After cloning the git repository, I ran `rosdep install --from-paths src --ignore-src -r -y` and this error appeared:

> ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
zed_nodelets: Cannot locate rosdep definition for [zed_interfaces]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

I ignored this error and I ran `catkin_make -DCMAKE_BUILD_TYPE=Release`, and then it appeared:

> -- Could NOT find zed_interfaces (missing: zed_interfaces_DIR)
-- Could not find the required component 'zed_interfaces'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "zed_interfaces"
  with any of the following names:

    zed_interfacesConfig.cmake
    zed_interfaces-config.cmake

>  Add the installation prefix of "zed_interfaces" to CMAKE_PREFIX_PATH or set
  "zed_interfaces_DIR" to a directory containing one of the above files.  If
  "zed_interfaces" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  zed-ros-wrapper/zed_nodelets/CMakeLists.txt:41 (find_package)

> -- Configuring incomplete, errors occurred!
See also "/home/isa/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/isa/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:320: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

And I am not able to continue. Today I've followed the same steps that I did a time ago and then no error appeared, but now I've found this error and I have done the same process.

Could you please tell me how to fix it?

I've seen that some days ago you updated the zed_nodelets, zed_ros_interfaces, etc. Maybe this could produce this error.

Myzhar, 2021-07-26:
Hi @joseandres15 please follow the updated instructions, giving attention to this:
`git clone --recursive https://github.com/stereolabs/zed-ros-wrapper.git`
If you have just cloned the repository without using the `--recursive` flag, then you can follow the [update instructions](https://github.com/stereolabs/zed-ros-wrapper#update-the-repository)

joseandres15, 2021-07-27:
It worked, thank you!

closed 2021-07-27 by Myzhar