stereolabs/zed-ros-wrapper #764: [Question] saveAreaMap

dreamonJure, 2021-07-27:
Hi, everyone! I encountered an issue when I used zed.saveAreaMap('my absolute path'). The problem is that no .area file was generated by the usage of this function. And then I used zed.getAreaExportState() to check state, it showed 'file exported successfully'. I was quite confused about it. So could anyone please give me some advice? Thank you very much.

Myzhar, 2021-07-27:
Hi @dreamonJure I will investigate this problem

Myzhar, 2021-07-27:
Related to this: https://github.com/stereolabs/zed-ros-wrapper/issues/547

dreamonJure, 2021-07-27:
Yeah, I've already seen that #547. However, my question is a little bit different. In #547, saveAreaMap returned FAILURE while created file to save map information. For my problem, I got SUCCESSFUL state returned by `zed.getAreaExportState()` without any file generated in the path I set.

closed 2021-07-28 by Myzhar