stereolabs/zed-ros-wrapper #769: [Question] Using nodelets to maintain rate of publishing of zed2 topics

kushal2507, 2021-08-14:
I am currently using a Zed2 camera to carry out obstacle detection with PCL. When I launch my camera I see that I am able to publish point clouds at 15 Hz but as I start subscribing multiple nodes to the same point cloud topic the publishing rate seems to drop. 

My question is -  Will using the nodelet manager help me in maintaining the original rate of publishing no matter the number of nodes which subscribe to it?

Also, soon I will have multiple cameras with multiple nodes subscribing to each camera topic. Would the nodelet manager be helpful in this case too?

I am currently using an i9 deca core processor (32 GB RAM) with a QuadroP1000(6 GB) Nvidia card on it . So I have more than enough computation power.

Thanking You,
Kushal

Myzhar, 2021-08-14:
Hi @kushal2507 
The ZED node has been written in order to take advantage of the zero-copy functionality of the ROS nodelet, so if you subscribe to a pointcloud topic in the same process running different nodes under the same nodelet manager surely you will improve the performances of your system.
You can start looking at the nodelet examples in the [zed-ros-examples Github repository ](https://github.com/stereolabs/zed-ros-examples)

kushal2507, 2021-08-17:
Thank you for your help!

closed 2021-08-17 by kushal2507