stereolabs/zed-ros-wrapper #772: [BUG] several frames of the image steam have left-right flipped rectification.

icsl-Jeon, 2021-08-19:
**Describe the bug**
The rectification image (`/zed2/zed_node/rgb/image_rect_color`) seems to be constructed with flipped left and right  in some case. For example, some frames look like the below:
[image].
The below is the very next frame from the above:
[image]
This kind of wrong rectification was observed in both ROS and ROS2 wrappers in your master release.

**Device information:**
 - OS: [Ubuntu] 
 - OS version: [20.04]
 - CPU: [amd ryzen 9]
 - GPU Nvidia:  [ RTX3070]

**To Reproduce**
`roslaunch zed_ros_wrapper zed2.launch`

Myzhar, 2021-08-19:
Hi @icsl-Jeon can you please verify if you notice the same problem using ZED_Explorer or ZED_Depth_Viewer?

icsl-Jeon, 2021-08-19:
Thank you for the reply. I've tested but they both included the weird frames also.

Myzhar, 2021-08-20:
Ok, that's what I wanted to know.
Please write an email to support@stereolabs.com to open a support ticket. This could be a problem with your USB port or with the camera cable. Our engineers will guide you to understand the cause of the issue.

icsl-Jeon, 2021-08-23:
I will send an email. Thank you!

closed 2021-08-23 by icsl-Jeon