stereolabs/zed-ros-wrapper #773: problem with recording SVO file

Pavol89, 2021-08-20:
I have problem with recording SVO file.

I use the command:
roslaunch zed_wrapper zed2.launch svo_file:="/home/pavol/Dokumenty/ZED/file.svo"

Error:
1626893633.623924920 INFO /zed2/zed_node [/home/palo/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:270(ZEDWrapperNodelet::onInit)] [topics: /rosout, /tf, /tf_static, /diagnostics] ZED connection -> INVALID SVO FILE

System:
-nvidia jetson nano 4gb
- ZED SDK 3.5.0
- ubuntu 18
- ROS melodic

thank you for any advice

Myzhar, 2021-08-20:
Hi @Pavol89 
the command you are using is to start the ZED node using an SVO as input.
To start recording an SVO you must call the `start_svo_recording` service while the node is running:
https://www.stereolabs.com/docs/ros/zed-node/#services

Pavol89, 2021-08-20:
Thank you for response.

I try this command, but the recording doesn't work. 

When I use only command `start_svo_recording` in new terminal.
Response is:
bash: start_svo_recording: command not found

When use  command  `rosservice call /zed-ros-interfaces/srv/start_svo_recording `
Responce is:
ERROR: Service [/zed-ros-interfaces/srv/start_svo_recording] in not available.

Pavol89, 2021-08-20:
From your previous answer: " command `roslaunch zed_wrapper zed2.launch svo_file:="/home/pavol/Dokumenty/ZED/file.svo"` are using is to start the ZED node using an SVO as input."

It means, that I can use this command for export all ROS topic from SVO file. 
There is a small number of images in the file bag that I export when using the command above. This also happens with the classic script `roslaunch zed_wrapper zed2.launch`. Could this be a problem with computing power?

First question: Is it possible to extract all ros themes using the first command?

Second question: When using zed-ros-wraper, missing images are computational performance problem? Or there is a better way to directly export SVO to AVI and select frames from AVI.

Myzhar, 2021-08-20:
Yes, computational power may be one source of the "small number of images": not recent GPU, drivers not updated, not fast CPU, small amount of RAM memory, slow mass memory, ...
What are the specifications of your host system?

Pavol89, 2021-08-20:
I used nvidia jetson nano:
GPU: 128-core Maxwell™ GPU
CPU: quad-core ARM® Cortex®-A57 CPU
Memory: 4GB 64-bit LPDDR4

ZEDfu doesn't work on my laptop. I think it's because of the graphics card (2GB NVIDIA GeForce GT 740M).
Other parameters are good:
CPU: Intel Core i7 3632QM @ 2.20GHz
RAM: 16GB

ZED Diagnostic show problem with graphhic card:
"Unable to detect your graphics card." Your graphics drivers need to be updated. Please download the latest drivers from.

I downloaded the latest available driver, but there is probably a problem with the fact that it is an older card.

What are the recommendations for the ZED SDK?
I found that the graphics card must be nvidia with virtual memory >= 2gb.
For a mobile application I must have a jetson xavier board [https://developer.nvidia.com/embedded/jetson-agx-xavier-developer-kit]?

closed 2022-04-19 by github-actions[bot]