stereolabs/zed-ros-wrapper #774: [BUG] In zed2.yaml, changing Param sensors_timestamp_sync = true breaks the transforms, and Error in Rviz " Fixed Frame Map does not Exist "

octobotics, 2021-08-29:
**Describe the bug**
In zed2.yaml, changing Param sensors_timestamp_sync = true breaks the transforms, and Error in Rviz " Fixed Frame Map does not Exist ", No TF are available , setting Param sensors_timestamp_sync = false , Error is gone.

**Device information:**
 - OS:  Ubuntu
 - OS version: 18.04
 - CPU: AMD  Ryzen 9 5900hx
 - GPU Nvidia: RTX3080
 - ZED SDK Version: 3.5
 - ROS Wrapper version: Latest Master

**To Reproduce**
Steps to reproduce the behavior:
1. Change Parameter " in Zed_model.YAML, Parameter = Sensors_timestamp_sync = True "
2. launch the camera  via roslaunch.
3.No TF published in RViz, Fixed Frame Map not available.
4. Change Parameter Sensors_timestamp_sync= False, Launch, Rviz Normal, TF published.

**Expected behavior**

**Screenshots**
[rviz ](https://drive.google.com/file/d/1eOL5TN2edk1r-PNUaFQChPE_YzzCdb83/view?usp=sharing)
**Additional context**
Add any other context about the problem here.

Myzhar, 2021-08-29:
What launch file are you using, `zed2.launch` or `display_zed2.launch`?

octobotics, 2021-08-29:
> What launch file are you using, `zed2.launch` or `display_zed2.launch`?

Hi ,

I've tried both the launch files. The error persists in both.

closed 2022-04-19 by github-actions[bot]