stereolabs/zed-ros-wrapper #776: Unable to detect ZED2 camera on Jetson Xavier [BUG]

jyang-cpr, 2021-08-30:
**Describe the bug**
Connected a ZED2 camera to the Jetson Xavier via the USB 3 port. 

Verified that HID and UVC modules are detected via `lsusb`:
```
Bus 002 Device 004: ID 2b03:f780  
Bus 002 Device 002: ID 0bda:0489 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 13d3:3549 IMC Networks 
Bus 001 Device 011: ID 2b03:f781  
Bus 001 Device 010: ID 0424:2512 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0bda:5489 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Output of `dmesg` after unplugging then replugging the camera:
```
[ 1531.555941] usb 2-3.1: Disable of device-initiated U1 failed.
[ 1531.557929] usb 2-3.1: Disable of device-initiated U2 failed.
[ 1532.400236] usb 1-2.1: USB disconnect, device number 4
[ 1532.669233] usb 1-2.1-port1: attempt power cycle
[ 1532.670933] usb 2-3.1: USB disconnect, device number 3
[ 1532.726471] usb 2-3: usb_suspend_both: status 0
[ 1532.726631] usb usb2: usb_suspend_both: status 0
[ 1532.982810] usb 1-2.1.2: USB disconnect, device number 5
[ 1533.166493] usb 1-2: usb_suspend_both: status 0
[ 1537.574411] usb 1-2.1: new high-speed USB device number 10 using tegra-xusb
[ 1537.594687] usb 1-2.1: New USB device found, idVendor=0424, idProduct=2512
[ 1537.594748] usb 1-2.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1537.595504] hub 1-2.1:1.0: USB hub found
[ 1537.595564] hub 1-2.1:1.0: 2 ports detected
[ 1537.886417] usb 1-2.1.2: new full-speed USB device number 11 using tegra-xusb
[ 1537.912439] usb 1-2.1.2: New USB device found, idVendor=2b03, idProduct=f781
[ 1537.912446] usb 1-2.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1537.912451] usb 1-2.1.2: Product: ZED-2 HID INTERFACE
[ 1537.912455] usb 1-2.1.2: Manufacturer: STEREOLABS
[ 1537.912458] usb 1-2.1.2: SerialNumber: 26488301
[ 1537.919722] hid-generic 0003:2B03:F781.0002: hidraw0: USB HID v1.11 Device [STEREOLABS ZED-2 HID INTERFACE] on usb-3610000.xhci-2.1.2/input0
[ 1538.002672] usb 2-3.1: new SuperSpeed USB device number 4 using tegra-xusb
[ 1538.023345] usb 2-3.1: New USB device found, idVendor=2b03, idProduct=f780
[ 1538.023363] usb 2-3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1538.023368] usb 2-3.1: Product: ZED 2
[ 1538.023372] usb 2-3.1: Manufacturer: Technologies, Inc.
```

Tried roslaunching `zed_no_tf.launch` and `zed2.launch`, getting the same error for both:
```
ROS_MASTER_URI=http://localhost:11311

process[zed2/zed2_state_publisher-1]: started with pid [10218]
process[zed2/zed_node-2]: started with pid [10219]
[ INFO] [1630356372.596185016]: Initializing nodelet with 2 worker threads.
[ INFO] [1630356372.722562105]: ********** Starting nodelet '/zed2/zed_node' **********
[ INFO] [1630356372.722813917]: SDK version : 3.5.3
[ INFO] [1630356372.723155394]: *** GENERAL PARAMETERS ***
[ INFO] [1630356372.726444818]:  * Camera Name			-> zed2
[ INFO] [1630356372.733562267]:  * Camera Resolution		-> HD720
[ INFO] [1630356372.736856747]:  * Camera Grab Framerate	-> 30
[ INFO] [1630356372.740814118]:  * Gpu ID			-> -1
[ INFO] [1630356372.743154248]:  * Camera ID			-> -1
[ INFO] [1630356372.745574764]:  * Verbose			-> DISABLED
[ INFO] [1630356372.752620755]:  * Camera Flip			-> DISABLED
[ INFO] [1630356372.764725957]:  * Self calibration		-> ENABLED
[ INFO] [1630356372.792453724]:  * Camera Model by param	-> zed2
[ INFO] [1630356372.792642783]: *** VIDEO PARAMETERS ***
[ INFO] [1630356372.804973492]:  * Image resample factor	-> 1
[ INFO] [1630356372.809949949]:  * Extrinsic param. frame	-> X RIGHT - Y DOWN - Z FWD
[ INFO] [1630356372.810128576]: *** DEPTH PARAMETERS ***
[ INFO] [1630356372.824630613]:  * Depth quality		-> PERFORMANCE
[ INFO] [1630356372.827076569]:  * Depth Sensing mode		-> STANDARD
[... 65 lines cut ...]

**Device information:**
 - OS: Ubuntu
 - OS version: 18.04.5 LTS
 - Kernel version: 4.9.140-tegra
 - CPU: Jetson Xavier 8-core ARM v8.2 64-bit CPU
 - GPU Nvidia:  512-core Volta GPU with Tensor Cores
 - ZED SDK Version: v3.5 (https://download.stereolabs.com/zedsdk/3.5/jp44/jetsons)
 - ROS Wrapper version:  latest master
 - CUDA version: 10.2.89
 - JetPack version: 4.4.1

**To Reproduce**
Steps to reproduce the behavior:
1. Plug ZED2 camera into Jetson Xavier via USB 3 port
2. Install ZED SDK version compatible for Jetson Xavier/CUDA 10.2/JetPack 4.4 from here: https://download.stereolabs.com/zedsdk/3.5/jp44/jetsons
3. Install and build `zed-ros-wrapper`
4. Run `roslaunch zed_no_tf.launch` or `roslaunch zed2.launch`
5. Obtain `ZED connection -> CAMERA NOT DETECTED` error

**Expected behavior**

**Screenshots**

**Additional context**

Myzhar, 2021-08-31:
Hi @jyang-cpr 
this seems to be a hardware problem. Please write an email to support@stereolabs.com attaching the report generated by the ZED Diagnostic tool. Please also test the camera with a different device (Desktop PC, laptop) to verify if the problem persists.
Thank you

jyang-cpr, 2021-08-31:
@Myzhar 

Thanks for the response. I tested the ZED2 camera on a Jetson Xavier AGX (not NX) running JetPack 4.6, and everything is working fine. Just doesn't seem to work on the Jetson NX with JetPack 4.4. I'll contact support@stereolabs.com with the diagnostics info, incase it is a hardware issue.

closed 2021-09-14 by Myzhar

jyang-cpr, 2021-09-21:
Just placing this information here incase it's helpful in the future:

I upgraded the Jetson NX to JetPack 4.6 (CUDA 10.2), and this error goes away. In fact, JetPack 4.6 is allowing the ZED2 camera to work reliably on the Jetson Xavier NX, Jetson Xavier AGX, and the Jetson Nano, as I tested all of these platforms last week. The only caveat is that because the ZED SDK has not been released for JP4.6 yet, I have been using the ZED SDK for JP4.4 (https://download.stereolabs.com/zedsdk/3.5/jp44/jetsons). I hope and assume that the ZED2 camera will continue to work reliably once the SDK for JP4.6 is released :)

Myzhar, 2021-09-21:
SDK v3.6 with JP4.6 support is in the final test phase. It will be released very soon