stereolabs/zed-ros-wrapper #777: Can I grab depth with sensing_mode FULL  in zed_wrapper module>

nonononentity, 2021-09-02:
Hello,
    When I run `roslaunch zed_wrapper zed2.launch` and modify the common.yaml with Sensing mode "FILL". I still get the depth of  STANDARD model.

Myzhar, 2021-09-02:
No, FILL mode is not good for robotics, that's why it's not available in the ZED ROS Wrapper. 
If for some reason you need it, you can add it by modifying the source code.

nonononentity, 2021-09-02:
Thank you for your answer. And I would also like to know another question:  When I use zed_wrapper , the  timestamps of "depth_registered" and "rgb/image_rect_color" collected by code are the actual timestamps or the timestamps in the grab operation(code line). In fact, when we tried to synchronize the remote camera with zed to create a binocular system, there was always a time difference about 200ms.

Myzhar, 2021-09-02:
The timestamp of the ROS topic is the timestamp provided by the ZED SDK and it's a mix of system timestamp and camera timestamp.
If you use different host systems you must synchronize the system clocks in order to have a synchronized timestamp.
You can follow this guide to use PTP for synchronization:
https://www.stereolabs.com/docs/video/multi-camera/

nonononentity, 2021-09-03:
Thank you.
  And I install PTP both  in my local pc and remote robot. When I check common 'ethtool' in remote robot, I find that remote robot doesn't support hardware timestamping and  even not provide full software PTP functionality,  it only has SOF_TIMESTAMPING_RX_SOFTWARE and SOF_TIMESTAMPING_SOFTWARE. 
  How do I synchronize time in this case？

Myzhar, 2021-09-14:
You can still use PTP even if it will provide a minor sync precision.

flabrosse, 2021-10-20:
Just to conclude on this, the FILL option is there and used.  As @Myzhar said, and as stated in the example config file, this might not be the right option for you (but it is for us).

The holes I was still getting despite the FILL mode were due to the confidence threshold.  I set it to 100 (accept everything) and the holes are gone.