stereolabs/zed-ros-wrapper #778: Reset SDK timestamp

jmaximoff, 2021-09-02:
Hi,

I'm using Zed ROS Wrapper for a robotics application on a system without a realtime clock (RTC), but which acts as an NTP client to a NTP server on the network (so that we can keep the clocks of many sensors and control systems synchronized). When the system starts up, the Zed node launches and latches an initial (incorrect) timestamp, and later even after NTP synchronization, continues to generate timestamps with respect to the initial incorrect timestamp. These timestamps appear to come from _sl_tools::slTime2Ros()_, so presumably this is all happening within the SDK.

The work-around I've been using is to detect the NTP synchronization and then force a restart of the Zed node, but of course that causes camera data to drop out for some seconds while the SDK and node reinitialize, so it's not ideal.

So, my question is whether there is any way to get the SDK to reset its baseline clock to the current system time without completely reinitializing it and incurring the long delay?

Myzhar, 2021-09-03:
The ZED SDK cannot work with the system clock making "big jumps", for this reason when it internally detects a big change the camera timestamp is slowly modified to adapt to the new system clock. You will notice that after a while the camera timestamp and the system clock will match after each change.
The only way to instantly adjust the camera timestamp to match a big system clock change is by closing it and re-opening.

jmaximoff, 2021-09-03:
Ok, thanks for the response. Can you provide a little more insight into what "after a while" means here? Like, what controls that rate and is it a function of the size of the time difference? I'm wondering if by adjusting some NTP parameters to make the system clock correction more gradual, I might speed up the ZED SDK clock convergence.

Myzhar, 2021-09-03:
@jmaximoff the deltaT applied is 4ms for each frame. So, the major is the time difference, the major is the time taken to get a matching timestamp.

jmaximoff, 2021-09-03:
Thanks -- that gives me enough to go on!

closed 2021-09-03 by jmaximoff