stereolabs/zed-ros-wrapper #779: Capture left and right RGB images on one device , generate depth on annother.

nonononentity, 2021-09-03:
Hello, 
     Can I capture left and right RGB images on a device without CUDA and generate depth maps on another device?(I remember seeing this somewhere in doc, but I can't find it.)

obraun-sl, 2021-09-03:
You can use ZED Explorer to generate a *.svo file that can be processed on another device with the ZED SDK + CUDA. 
ZED Explorer can be used without CUDA, just the compression mode will only be lossless (cpu)

nonononentity, 2021-09-03:
If we just capture left and right images in a unsupported system(zed just be considered a common camera device), can I generate depth by myself with products' SDKs?

Myzhar, 2021-09-03:
No, that's not possible. You must record an SVO using ZED Explorer (it does not use CUDA so you can run it on every system) and use that SVO as input for the SDK on other CUDA-capable systems

nonononentity, 2021-09-03:
ok,  thank you for your answer!

Myzhar, 2021-09-03:
You can close the issue if you do not need other information.
For the next questions not strictly related to the ZED ROS Wrapper and robotics, I suggest you write a post on the Stereolabs community forum: https://community.stereolabs.com/

closed 2021-09-06 by nonononentity