stereolabs/zed-ros-wrapper #781: ZED topic rate drops

kushal2507, 2021-09-06:
I am running 2 PCs with two zed2 cameras on each. The computers are connected to each other through a high speed gigabit network. So essentially, one pc is the master and the other one is the slave.

The ZED camera on the host pc publishes topics at the specified rate but the same topic when viewed on the other pc gives me half the rate. At this moment I have not even started subscribing to these topics.

I use rostopic hz to view the publishing rate. 

PC specs (identical for both pcs) - 
intel i7 11th gen
64gb ram
nvidia rtx 2060 6GB

Myzhar, 2021-09-06:
Hi @kushal2507 
can you provide more information about the topic that you are subscribing to? 
This seems a bandwidth problem not strictly related to the ZED Wrapper...

kushal2507, 2021-09-06:
I'm subscribing to the pointcloud topic. 
We have a gigabit switch and a gigabit router for our networking.

Myzhar, 2021-09-06:
What resolution?
You can also try `$ rostopic bw <topic>` to monitor the bandwidth

kushal2507, 2021-09-06:
RGB -  720 p
Depth - Performance
I am trying to publish images and pointcloud at 15Hz.

I tried rostopic bw /zed2/zed_node/point_cloud/cloud_registered

Host PC -  average: 221.42MB/s
	mean: 14.75MB min: 14.75MB max: 14.75MB window: 12
Other PC - 118.52MB/s
	mean: 14.75MB min: 14.75MB max: 14.75MB window: 54

Myzhar, 2021-09-06:
@kushal2507 
221.42 MB/s is more than 1 Gbit/s, it's ~1.73 Gbit/s
On the remote PC indeed you get 118.52 MB/s that's ~1Gbit/s (a bit lower, but there is other traffic on the network)

kushal2507, 2021-09-06:
Remote PC : average: 58.98MB/s
	mean: 3.69MB min: 3.69MB max: 3.69MB window: 14

Host PC : average: 57.13MB/s
	mean: 3.69MB min: 3.69MB max: 3.69MB window: 30

I get the above results if I check the left/image_rect_color image topic. The rate of publishing is also constant at 15 Hz.

The descripancy is happening only with the point cloud topic from what I can see. 

What would you suggest I do? Look into a new network switch? Or are there any settings I can change within ZED or ROS for that matter?

Myzhar, 2021-09-06:
Can you copy and paste here your `common.yaml` file?

kushal2507, 2021-09-07:
# params/common.yaml
# Common parameters to Stereolabs ZED and ZED mini cameras
---

# Dynamic parameters cannot have a namespace
brightness:                 4                                   # Dynamic
contrast:                   4                                   # Dynamic
hue:                        0                                   # Dynamic
saturation:                 4                                   # Dynamic
sharpness:                  4                                   # Dynamic
gamma:                      8                                   # Dynamic - Requires SDK >=v3.1
auto_exposure_gain:         true                                # Dynamic
gain:                       100                                 # Dynamic - works only if `auto_exposure_gain` is false
exposure:                   100                                 # Dynamic - works only if `auto_exposure_gain` is false
auto_whitebalance:          true                                # Dynamic
whitebalance_temperature:   42                                  # Dynamic - works only if `auto_whitebalance` is false
depth_confidence:           50                                  # Dynamic
depth_texture_conf:         100                                 # Dynamic
pub_frame_rate:             15.0                                # Dynamic - frequency of publishing of video and depth data
point_cloud_freq:           15.0                                # Dynamic - frequency of the pointcloud publishing (equal or less to `grab_frame_rate` value)

general:
    camera_name:                zed                             # A name for the camera (can be different from camera model and node name and can be overwritten by the launch file)
    zed_id:                     0
    serial_number:              0
    resolution:                 2                               # '0': HD2K, '1': HD1080, '2': HD720, '3': VGA
    grab_frame_rate:            30                              # Frequency of frame grabbing for internal SDK operations
    gpu_id:                     -1
    base_frame:                 'base_link'                     # must be equal to the frame_id used in the URDF file
    verbose:                    false                           # Enable info message by the ZED SDK
    svo_compression:            2                               # `0`: LOSSLESS, `1`: AVCHD, `2`: HEVC
    self_calib:                 true                            # enable/disable self calibration at starting
    camera_flip:                false

video:
    img_downsample_factor:      1.0                             # Resample factor for images [0.01,1.0] The SDK works with native image sizes, but publishes rescaled image.
    extrinsic_in_camera_frame:  true                            # if `false` extrinsic parameter in `camera_info` will use ROS native frame (X FORWARD, Z UP) instead of the camera frame (Z FORWARD, Y DOWN) [`true` use old behavior as for version < v3.1]

depth:
    quality:                    2                               # '0': NONE, '1': PERFORMANCE, '2': QUALITY, '3': ULTRA
    sensing_mode:               0                               # '0': STANDARD, '1': FILL (not use FILL for robotic applications)
    depth_stabilization:        1                               # `0`: disabled, `1`: enabled
    openni_depth_mode:          false                           # 'false': 32bit float meters, 'true': 16bit uchar millimeters
    depth_downsample_factor:    1.0                             # Resample factor for depth data matrices [0.01,1.0] The SDK works with native data sizes, but publishes rescaled matrices (depth map, point cloud, ...)

pos_tracking:
    pos_tracking_enabled:       true                            # True to enable positional tracking from start
    publish_tf:                 true                            # publish `odom -> base_link` TF
    publish_map_tf:             true                            # publish `map -> odom` TF
    map_frame:                  'map'                           # main frame
    odometry_frame:             'odom'                          # odometry frame
    area_memory_db_path:        'zed_area_memory.area'          # file loaded when the node starts to restore the "known visual features" map. 
    save_area_memory_db_on_exit: false                          # save the "known visual features" map when the node is correctly closed to the path indicated by `area_memory_db_path`
    area_memory:                true                            # Enable to detect loop closure
    floor_alignment:            false                           # Enable to automatically calculate camera/floor offset
    initial_base_pose:          [0.0,0.0,0.0, 0.0,0.0,0.0]      # Initial position of the `base_frame` -> [X, Y, Z, R, P, Y]
    init_odom_with_first_valid_pose: true                       # Enable to initialize the odometry with the first valid pose
    path_pub_rate:              2.0                             # Camera trajectory publishing frequency
    path_max_count:             -1                              # use '-1' for unlimited path size
    two_d_mode:                 false                           # Force navigation on a plane. If true the Z value will be fixed to "fixed_z_value", roll and pitch to zero
    fixed_z_value:              0.00                            # Value to be used for Z coordinate if `two_d_mode` is true

mapping:
    mapping_enabled:            false                           # True to enable mapping and fused point cloud pubblication
    resolution:                 0.05                            # maps resolution in meters [0.01f, 0.2f]
    max_mapping_range:          -1                              # maximum depth range while mapping in meters (-1 for automatic calculation) [2.0, 20.0]
    fused_pointcloud_freq:      1.0                             # frequency of the publishing of the fused colored point cloud

Myzhar, 2021-09-07:
OK, like I supposed you are using the default parameters, so it's correct that the topic rate is ~15Hz.
There is more than a parameter to set the topic rate:

* `grab_frame_rate`: it's the grabbing rate of the ZED SDK. This can be higher than the publishing rate
* `pub_frame_rate`: publishing rate of the RGB and depth topics. This can be equal/lower than `grab_frame_rate`. This is a dynamic parameter
* `point_cloud_freq`: publishing rate of the point cloud topic. This can be equal/lower than `grab_frame_rate`. This is a dynamic parameter

You can get the full description of all the parameters on the online documentation:
https://www.stereolabs.com/docs/ros/zed-node/#zed-parameters

Regarding the point cloud, if the bandwidth is not enough you can switch to `openni mode` setting the `openni_depth_mode` to true. In this way, 16-bit unsigned int values will be used for the points coordinates instead of 32-bit floating-point values.
Be aware that openni mode will provide points coordinate in millimeters and not in meters.
You can read more in the official [ROS REP108](https://www.ros.org/reps/rep-0118.html) document

closed 2022-04-19 by github-actions[bot]