stereolabs/zed-ros-wrapper #783: [BUG]

peredwardsson, 2021-09-09:
**Describe the bug**
Zed node does not publish a bunch of images, seemingly because parameters which need (by image_transport) to be dynamic are not. Startup reveals the following:
```
[zed_wrapper-1] 1631188736.977301279 [sensors.zed_node] [INFO] *** PUBLISHED TOPICS ***
[zed_wrapper-1] 1631188737.019904134 [sensors.zed_node] [WARN] zed_node.rgb.image_rect_color.format is not a dynamic parameter
[zed_wrapper-1] 1631188737.023878720 [sensors.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.rgb.image_rect_color.format' could not be set: zed_node.rgb.image_rect_color.format is not a dynamic parameter
[zed_wrapper-1] 
[zed_wrapper-1] 1631188737.049994140 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/rgb/image_rect_color
[zed_wrapper-1] 1631188737.056585625 [sensors.zed_node] [WARN] zed_node.rgb.image_rect_gray.format is not a dynamic parameter
[zed_wrapper-1] 1631188737.059316618 [sensors.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.rgb.image_rect_gray.format' could not be set: zed_node.rgb.image_rect_gray.format is not a dynamic parameter
[zed_wrapper-1] 
[zed_wrapper-1] 1631188737.065476151 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/rgb/image_rect_gray
[zed_wrapper-1] 1631188737.067301315 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/rgb/camera_info
[zed_wrapper-1] 1631188737.071811193 [sensors.zed_node] [WARN] zed_node.rgb_raw.image_raw_color.format is not a dynamic parameter
[zed_wrapper-1] 1631188737.074358548 [sensors.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.rgb_raw.image_raw_color.format' could not be set: zed_node.rgb_raw.image_raw_color.format is not a dynamic parameter
[zed_wrapper-1] 
[zed_wrapper-1] 1631188737.081491598 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/rgb_raw/image_raw_color
[zed_wrapper-1] 1631188737.086181177 [sensors.zed_node] [WARN] zed_node.rgb_raw.image_raw_gray.format is not a dynamic parameter
[zed_wrapper-1] 1631188737.088591845 [sensors.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.rgb_raw.image_raw_gray.format' could not be set: zed_node.rgb_raw.image_raw_gray.format is not a dynamic parameter
[zed_wrapper-1] 
[zed_wrapper-1] 1631188737.094256860 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/rgb_raw/image_raw_gray
[zed_wrapper-1] 1631188737.095788294 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/rgb_raw/camera_info
[zed_wrapper-1] 1631188737.099922418 [sensors.zed_node] [WARN] zed_node.left.image_rect_color.format is not a dynamic parameter
[zed_wrapper-1] 1631188737.102787825 [sensors.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.left.image_rect_color.format' could not be set: zed_node.left.image_rect_color.format is not a dynamic parameter
[zed_wrapper-1] 
[zed_wrapper-1] 1631188737.111755832 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/left/image_rect_color
[zed_wrapper-1] 1631188737.117028355 [sensors.zed_node] [WARN] zed_node.left.image_rect_gray.format is not a dynamic parameter
[zed_wrapper-1] 1631188737.120104697 [sensors.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.left.image_rect_gray.format' could not be set: zed_node.left.image_rect_gray.format is not a dynamic parameter
[zed_wrapper-1] 
[zed_wrapper-1] 1631188737.128240675 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/left/image_rect_gray
[zed_wrapper-1] 1631188737.130506112 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/left/camera_info
[zed_wrapper-1] 1631188737.136260992 [sensors.zed_node] [WARN] zed_node.left_raw.image_raw_color.format is not a dynamic parameter
[zed_wrapper-1] 1631188737.139167396 [sensors.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.left_raw.image_raw_color.format' could not be set: zed_node.left_raw.image_raw_color.format is not a dynamic parameter
[zed_wrapper-1] 
[zed_wrapper-1] 1631188737.143184131 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/left_raw/image_raw_color
[zed_wrapper-1] 1631188737.149581547 [sensors.zed_node] [WARN] zed_node.left_raw.image_raw_gray.format is not a dynamic parameter
[zed_wrapper-1] 1631188737.153322252 [sensors.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.left_raw.image_raw_gray.format' could not be set: zed_node.left_raw.image_raw_gray.format is not a dynamic parameter
[zed_wrapper-1] 
[zed_wrapper-1] 1631188737.175626239 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/left_raw/image_raw_gray
[zed_wrapper-1] 1631188737.177797457 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/left_raw/camera_info
[zed_wrapper-1] 1631188737.183127874 [sensors.zed_node] [WARN] zed_node.right.image_rect_color.format is not a dynamic parameter
[zed_wrapper-1] 1631188737.188292417 [sensors.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.right.image_rect_color.format' could not be set: zed_node.right.image_rect_color.format is not a dynamic parameter
[zed_wrapper-1] 
[zed_wrapper-1] 1631188737.199475774 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/right/image_rect_color
[zed_wrapper-1] 1631188737.206184201 [sensors.zed_node] [WARN] zed_node.right.image_rect_gray.format is not a dynamic parameter
[zed_wrapper-1] 1631188737.210531469 [sensors.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.right.image_rect_gray.format' could not be set: zed_node.right.image_rect_gray.format is not a dynamic parameter
[zed_wrapper-1] 
[zed_wrapper-1] 1631188737.221645699 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/right/image_rect_gray
[zed_wrapper-1] 1631188737.228560933 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/right/camera_info
[zed_wrapper-1] 1631188737.237767302 [sensors.zed_node] [WARN] zed_node.right_raw.image_raw_color.format is not a dynamic parameter
[zed_wrapper-1] 1631188737.243474369 [sensors.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.right_raw.image_raw_color.format' could not be set: zed_node.right_raw.image_raw_color.format is not a dynamic parameter
[zed_wrapper-1] 
[zed_wrapper-1] 1631188737.253182746 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/right_raw/image_raw_color
[zed_wrapper-1] 1631188737.260981182 [sensors.zed_node] [WARN] zed_node.right_raw.image_raw_gray.format is not a dynamic parameter
[zed_wrapper-1] 1631188737.261962571 [sensors.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.right_raw.image_raw_gray.format' could not be set: zed_node.right_raw.image_raw_gray.format is not a dynamic parameter
[zed_wrapper-1] 
[zed_wrapper-1] 1631188737.268407385 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/right_raw/image_raw_gray
[zed_wrapper-1] 1631188737.268542184 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/right_raw/camera_info
[zed_wrapper-1] 1631188737.274499935 [sensors.zed_node] [WARN] zed_node.depth.depth_registered.format is not a dynamic parameter
[zed_wrapper-1] 1631188737.276130037 [sensors.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.depth.depth_registered.format' could not be set: zed_node.depth.depth_registered.format is not a dynamic parameter
[zed_wrapper-1] 
[zed_wrapper-1] 1631188737.283654651 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/depth/depth_registered
[zed_wrapper-1] 1631188737.283975262 [sensors.zed_node] [INFO] Advertised on topic: /sensors/zed_node/depth/camera_info
[zed_wrapper-1] 1631188737.288408140 [sensors.zed_node] [WARN] zed_node.stereo.image_rect_color.format is not a dynamic parameter
[zed_wrapper-1] 1631188737.289747393 [sensors.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.stereo.image_rect_color.format' could not be set: zed_node.stereo.image_rect_color.format is not a dynami [... cut ...]
 - GPU Nvidia:  See above
 - ZED SDK Version: 3.5 (from https://download.stereolabs.com/zedsdk/3.5/jp45/jetsons)
 - ROS Wrapper version: commit 5fc188de1353f9254be1bf825b82127c282c0e8b
 - ROS2 version: Foxy

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Zed2 to Jetson via USB.
2. Start nodes with default config

**Expected behavior**
Node starts publishing sensor data, all manners of imagery and pictures.

**Additional information**
Image transport built from source with branch `foxy-devel`

Myzhar, 2021-09-09:
@peredwardsson there is something wrong. This is the Github repository of the ZED Wrapper for ROS1, it supports ROS melodic and ROS noetic.
For ROS2 Foxy you must use the following repository
https://github.com/stereolabs/zed-ros2-wrapper
Please also read the "[readme](https://github.com/stereolabs/zed-ros2-wrapper/blob/master/README.md)" file with attention because there are [known issues](https://github.com/stereolabs/zed-ros2-wrapper/blob/master/README.md#known-issues) with the `image_transport` package in ROS2 Foxy.

peredwardsson, 2021-09-09:
Apologies, will repost.

closed 2021-09-09 by peredwardsson