stereolabs/zed-ros-wrapper #784: 11.0 cuda not work well with the newest zed sdk?

shaoxh, 2021-09-17:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I have only cuda 11.0 in my device. but when I run the zed_diagnostics tools, It told a warning multiple version of CUDA. 
I have no idea with this issue. 
this problem has kept me away from using object detection function in zed2.
because the first time I rosrun zed, error like [ERROR] [1631870279.657774596]: Object detection error: CORRUPTED SDK INSTALLATION showed up.

### Steps to Reproduce

1. install zed2 sdk 3.4.5
2. run zed_diagnotics tool
3. roslaunch zed_wrapper zed2.launch

...

### Expected Result

no exception

### Actual Result

[ERROR] [1631870279.657774596]: Object detection error: CORRUPTED SDK INSTALLATION

### ZED Camera model

ZED2

### Environment

```shell
ubuntu 18.04, ros melodic, gpu3060 with cuda 11.0, zed2 sdk 3.5.4
```

Myzhar, 2021-09-17:
Hi @shaoxh 
When you update the ZED SDK it's a good practice to also clean the ROS workspace cache in order to force the compiler to recompile everything and link to the new ZED libraries.
Please delete the `devel` and the `build` folders in the root of your catkin workspace, then run a new `catkin_make` command.

shaoxh, 2021-09-17:
Hi @Myzhar 
I actually did what you said, but still not working.
Did deleting /usr/local/zed mean remove all former SDK totally ? Is there anything leaving not to do ?

Myzhar, 2021-09-17:
Yes, everything is in `/usr/local/zed`, so if you delete this folder you fully remove an old installation of the ZED SDK.
Can I suggest you try to upgrade to CUDA 11.1? I know that CUDA 11.0 can cause some problems.

shaoxh, 2021-09-17:
Ok I'll do it and check if  this error still there @Myzhar . QAQ. 
To be mentioned, when running the zed diagnotics tools, why it tells me a warning of multiple version of cuda? 
Do you have any idea?

Myzhar, 2021-09-17:
We are investigating that. It happens in some particular conditions and we must fix it.

shaoxh, 2021-09-17:
Hi @Myzhar ,
I have upgrading it to 11.1, it is not working. Then I install a 10.2 cuda, and this problem not show but a message like  [INFO] [1631879402.696878102]: ZED connection -> NO GPU COMPATIBLE shows up.
In this condition, roslaunch can never work.
Could you help with it?

Myzhar, 2021-09-17:
You must clean everything before upgrading/downgrading, to be sure that compatibilities are respected.
```
sudo apt purge nvidia-*
sudo apt purge cuda-*
```
then reboot and next install the ZED SDK which will guide you installing the correct Nvidia driver and CUDA versions.

shaoxh, 2021-09-17:
Thanks for your kind support. It's a long story to tell.

My GPU(nvidia 3060) could not work with cuda 10.2, so I removed all the Nvidia driver & cuda. 
Then I installed nvidia driver first. It is a 460 version driver.
As you informed me that zed SDK has its own program to install a respected cuda, I did not install cuda on my own this time,  but directly used the ZED SDK run file. 
It is a SDK with CUDA 11.0.
I chose yes to make it install this respected CUDA when installing ZED SDK.
Things should be OK, because I let ZED SDK deal with the CUDA thing.
However, this error, Object detection error: CORRUPTED SDK INSTALLATION, showed again. 

Now I am pretty sure there is a bug in your SDK to work with this kind of Nvidia driver and its corresponding CUDA toolkit.

So far I have to give up with the object detection feature, frustrating....

closed 2022-04-19 by github-actions[bot]