stereolabs/zed-ros-wrapper #785: Object Detection Error Zed2i

pouyashiri, 2021-09-23:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi!
I have a Zed2i camera and I am using Ubuntu 20.04, latest SDK (3.5.5), CUDA 11.1 and ROS noetic. When I start the zed2i node with object detection enabled, I get the following error:
`[ INFO] [1632417725.739049963]: *** Starting Object Detection ***
[ERROR] [1632417725.739090440]: Object detection error: MOTION SENSORS REQUIRED`
I appreciate your help and support. 
Have a great day

### Steps to Reproduce

1. Enable object detection.
2. Start the zed2i ros node.

### Expected Result

Getting object detection to work.

### Actual Result

Got the following error:
Object detection error: MOTION SENSORS REQUIRED

### ZED Camera model

ZED2i

### Environment

```shell
Ubunbu 20.04
CUDA 11.1
SDK 3.5.5
ROS noetic
```

pouyashiri, 2021-10-12:
Any comments on this?

closed 2022-04-19 by github-actions[bot]

l43lu, 2023-05-11:
Hello, currently experiencing the same issue. Are there any updates on this?

Shubham-2302, 2024-01-21:
Did you get it working?