stereolabs/zed-ros-wrapper #792: Can't compile zed-ros-wrapper

Heintsing, 2021-11-03:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When I follow the zed-ros-wrapper install instruction:
  $ cd ~/catkin_ws/src
  $ git clone --recursive https://github.com/stereolabs/zed-ros-wrapper.git
  $ cd ../
  $ rosdep install --from-paths src --ignore-src -r -y
  $ catkin_make -DCMAKE_BUILD_TYPE=Release
  $ source ./devel/setup.bash
There is one error occured.

### Steps to Reproduce

  $ cd ~/catkin_ws/src
  $ git clone --recursive https://github.com/stereolabs/zed-ros-wrapper.git
  $ cd ../
  $ rosdep install --from-paths src --ignore-src -r -y
  $ catkin_make -DCMAKE_BUILD_TYPE=Release1.

### Expected Result

compile success.

### Actual Result

[  0%] Built target tf2_msgs_generate_messages_nodejs
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target tf2_msgs_generate_messages_lisp
[  0%] Built target tf2_msgs_generate_messages_py
[  0%] Built target nodelet_generate_messages_lisp
[ 18%] Built target zed_interfaces_generate_messages_cpp
[ 36%] Built target zed_interfaces_generate_messages_lisp
[ 56%] Built target zed_interfaces_generate_messages_py
[ 94%] Built target zed_interfaces_generate_messages_nodejs
[ 94%] Built target zed_interfaces_generate_messages_eus
[ 94%] Built target zed_interfaces_generate_messages
[ 96%] Built target zed_wrapper_node
[ 97%] Building CXX object zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o
/home/zht/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘virtual void zed_nodelets::ZEDWrapperNodelet::onInit()’:
/home/zht/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:339:43: error: ‘ZED2i’ is not a member of ‘sl::MODEL’
   else if (mZedRealCamModel == sl::MODEL::ZED2i)
                                           ^~~~~
/home/zht/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:550:79: error: ‘ZED2i’ is not a member of ‘sl::MODEL’
     if (mZedRealCamModel == sl::MODEL::ZED2 || mZedRealCamModel == sl::MODEL::ZED2i)
                                                                               ^~~~~
/home/zht/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘void zed_nodelets::ZEDWrapperNodelet::readParameters()’:
/home/zht/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:698:35: error: ‘ZED2i’ is not a member of ‘sl::MODEL’
     mZedUserCamModel = sl::MODEL::ZED2i;
                                   ^~~~~
/home/zht/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘bool zed_nodelets::ZEDWrapperNodelet::start_obj_detect()’:
/home/zht/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:1510:77: error: ‘ZED2i’ is not a member of ‘sl::MODEL’
   if (mZedRealCamModel != sl::MODEL::ZED2 && mZedRealCamModel != sl::MODEL::ZED2i)
                                                                             ^~~~~
/home/zht/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘void zed_nodelets::ZEDWrapperNodelet::publishSensData(ros::Time)’:
/home/zht/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:3100:77: error: ‘ZED2i’ is not a member of ‘sl::MODEL’
   if (mZedRealCamModel == sl::MODEL::ZED2 || mZedRealCamModel == sl::MODEL::ZED2i)
                                                                             ^~~~~
/home/zht/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:3193:77: error: ‘ZED2i’ is not a member of ‘sl::MODEL’
   if (mZedRealCamModel == sl::MODEL::ZED2 || mZedRealCamModel == sl::MODEL::ZED2i)
                                                                             ^~~~~
/home/zht/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘void zed_nodelets::ZEDWrapperNodelet::callback_updateDiagnostic(diagnostic_updater::DiagnosticStatusWrapper&)’:
/home/zht/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:4395:77: error: ‘ZED2i’ is not a member of ‘sl::MODEL’
   if (mZedRealCamModel == sl::MODEL::ZED2 || mZedRealCamModel == sl::MODEL::ZED2i)
                                                                             ^~~~~
/home/zht/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘bool zed_nodelets::ZEDWrapperNodelet::on_start_object_detection(zed_interfaces::start_object_detection::Request&, zed_interfaces::start_object_detection::Response&)’:
/home/zht/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:4734:77: error: ‘ZED2i’ is not a member of ‘sl::MODEL’
   if (mZedRealCamModel != sl::MODEL::ZED2 && mZedRealCamModel != sl::MODEL::ZED2i)
                                                                             ^~~~~
zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/build.make:86: recipe for target 'zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:2520: recipe for target 'zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed

### ZED Camera model

ZED2

### Environment

```shell
CPU: Ii7
GPU NVIDA 2070
CUDA: 10.0
SDK: 3.4.2
```

Myzhar, 2021-11-03:
The current version of the ZED ROS Wrapper requires at least the ZED SDK v3.5 as reported in the requirements:
https://github.com/stereolabs/zed-ros-wrapper#prerequisites

Heintsing, 2021-11-03:
> The current version of the ZED ROS Wrapper requires at least the ZED SDK v3.5 as reported in the requirements: https://github.com/stereolabs/zed-ros-wrapper#prerequisites

Thank you very much.

closed 2022-04-19 by github-actions[bot]